RT-Thread/rt-thread #5323: ulog组件空指针与空字符混用的bug

lixiaolong1997, 2021-12-02:
1、ulog_output_to_all_backend函数中若后端不支持颜色或者不是raw，会进入else分支，else中使用使用rt_strlen计算各个输出等级颜色字符串的长度。作者想当然认为color_output_info中存储的是非空字符串，然而color_output_info中是字符串指针，有可能是NULL空指针，而非空字符串，这里误用空指针和空字符串，势必错误。
   if (backend->support_color || is_raw)
        {
            backend->output(backend, level, tag, is_raw, log, size);
        }
        else
        {
            /* recalculate the log start address and log size when backend not supported color */
            **rt_size_t color_info_len = rt_strlen(color_output_info[level]), output_size = size;**
            if (color_info_len)
            {
                rt_size_t color_hdr_len = rt_strlen(CSI_START) + color_info_len;

                log += color_hdr_len;
                output_size -= (color_hdr_len + (sizeof(CSI_END) - 1));
            }
            backend->output(backend, level, tag, is_raw, log, output_size);
        }
2、ulog_formater中用到color_output_info时，又认为存储的对象是指针.....，向后不一致。

#ifdef ULOG_USING_COLOR
    /* add CSI start sign and color info */
    **if (color_output_info[level])**
    {
        log_len += ulog_strcpy(log_len, log_buf + log_len, CSI_START);
        log_len += ulog_strcpy(log_len, log_buf + log_len, color_output_info[level]);
    }
#endif /* ULOG_USING_COLOR */
3、
#ifdef ULOG_USING_COLOR
/* color output info */
static const char * const color_output_info[] =
{
        ULOG_COLOR_ASSERT,
        NULL,
        NULL,
        ULOG_COLOR_ERROR,
        ULOG_COLOR_WARN,
        NULL,
        ULOG_COLOR_INFO,
        ULOG_COLOR_DEBUG,
};
#endif /* ULOG_USING_COLOR */
4、空指针、空字符串傻傻分不清，还写系统，只能哈哈了。
5、修复很简单，自己搞定。

thewon86, 2021-12-03:
const char *  就是指针啊
`rt_strlen(color_output_info[level])` 可以这么用，可能是因为 rt_strlen(NULL) 的返回结果是 0 吧。后面运行结果和先判断指针是否为空是一样的。
但是，先判断指针后使用是个好习惯

liukangcc, 2022-03-15:
[image]

 没看出有啥毛病 @mysterywolf

willianchanlovegithub, 2022-03-15:
> rt_strlen(NULL)

willianchanlovegithub, 2022-03-15:
> const char * 就是指针啊 `rt_strlen(color_output_info[level])` 可以这么用，可能是因为 rt_strlen(NULL) 的返回结果是 0 吧。后面运行结果和先判断指针是否为空是一样的。 但是，先判断指针后使用是个好习惯

rt_strlen(NULL) 不可以这么使用，会访问 0 地址报错。你进去看看 rt_strlen 源码就能发现了：for 循环中的判断条件是 `*sc != '\0'`，rt_strlen(NULL) 相当于 *0。

closed 2022-03-15 by willianchanlovegithub

willianchanlovegithub, 2022-03-15:
`color_output_info[level]`  这个东西，在 ulog 中使用之前，不每次都有个 if 去判断是否非空了嘛