RT-Thread/rt-thread #5338: 使用新唐M2354芯片进行CAN通信时，RTT版本为4.0.4，调用rt_device_write(can_dev, 0, &msg, sizeof(msg))时会卡住，好像是一直在_pm_device_check_idle函数中死循环。

Leoxuyu, 2021-12-06:


cazure, 2021-12-07:
发论坛问吧，估计是你没接CAN设备，没收到ACK，或者波特率错误之类的问题

Leoxuyu, 2021-12-07:
### 问题描述
在使用RTT 4.0.4版本代码，CAN0使用中断发送和中断接收方式打开。使用CAN0接口发送一帧数据，发现数据发送完成后，无法进入中断，无法触发接收中断。导致发送完毕卡死在rt_completion_wait(&(tx_fifo->buffer[no].completion), RT_WAITING_FOREVER);另外别和我说是硬件没有连接或者波特率不对，肯定不是这个问题，因为CAN总线上连着ZLG USBCAN II，可以收到发送出来的一帧报文，同时也没有主动错误或者被动错误。另外接收数据也有问题，CAN卡发送数据后无法接收并且打印数据。

### 测试代码
'''C
////////////////////////////////////////////
/*
 * 程序清单：这是一个 CAN 设备使用例程
 * 例程导出了 can_sample 命令到控制终端
 * 命令调用格式：can_sample can1
 * 命令解释：命令第二个参数是要使用的 CAN 设备名称，为空则使用默认的 CAN 设备
 * 程序功能：通过 CAN 设备发送一帧，并创建一个线程接收数据然后打印输出。
*/

#include <rtthread.h>
#include "rtdevice.h"

#define CAN_DEV_NAME       "can0"      /* CAN 设备名称 */

static struct rt_semaphore rx_sem;     /* 用于接收消息的信号量 */
static rt_device_t can_dev;            /* CAN 设备句柄 */

/* 接收数据回调函数 */
static rt_err_t can_rx_call(rt_device_t dev, rt_size_t size)
{
    /* CAN 接收到数据后产生中断，调用此回调函数，然后发送接收信号量 */
    rt_sem_release(&rx_sem);

    return RT_EOK;
}

static void can_rx_thread(void *parameter)
{
    int i;
    rt_err_t res;
    struct rt_can_msg rxmsg = {0};

    /* 设置接收回调函数 */
    rt_device_set_rx_indicate(can_dev, can_rx_call);

#ifdef RT_CAN_USING_HDR
    struct rt_can_filter_item items[5] =
    {
        RT_CAN_FILTER_ITEM_INIT(0x100, 0, 0, 1, 0x700, RT_NULL, RT_NULL), /* std,match ID:0x100~0x1ff，hdr 为 - 1，设置默认过滤表 */
        RT_CAN_FILTER_ITEM_INIT(0x300, 0, 0, 1, 0x700, RT_NULL, RT_NULL), /* std,match ID:0x300~0x3ff，hdr 为 - 1 */
        RT_CAN_FILTER_ITEM_INIT(0x211, 0, 0, 1, 0x7ff, RT_NULL, RT_NULL), /* std,match ID:0x211，hdr 为 - 1 */
        RT_CAN_FILTER_STD_INIT(0x486, RT_NULL, RT_NULL),                  /* std,match ID:0x486，hdr 为 - 1 */
        {0x555, 0, 0, 1, 0x7ff, 7,}                                       /* std,match ID:0x555，hdr 为 7，指定设置 7 号过滤表 */
    };
    struct rt_can_filter_config cfg = {5, 1, items}; /* 一共有 5 个过滤表 */
    /* 设置硬件过滤表 */
    res = rt_device_control(can_dev, RT_CAN_CMD_SET_FILTER, &cfg);
    RT_ASSERT(res == RT_EOK);
#endif

    while (1)
    {
        /* hdr 值为 - 1，表示直接从 uselist 链表读取数据 */
        rxmsg.hdr = -1;
        /* 阻塞等待接收信号量 */
        rt_sem_take(&rx_sem, RT_WAITING_FOREVER);
        /* 从 CAN 读取一帧数据 */
        rt_device_read(can_dev, 0, &rxmsg, sizeof(rxmsg));
        /* 打印数据 ID 及内容 */
        rt_kprintf("ID:%x", rxmsg.id);
        for (i = 0; i < 8; i++)
        {
            rt_kprintf("%2x", rxmsg.data[i]);
        }

        rt_kprintf("\n");
    }
}

int can_sample(int argc, char *argv[])
{
    struct rt_can_msg msg = {0};
    rt_err_t res;
    rt_size_t  size;
    rt_thread_t thread;
    char can_name[RT_NAME_MAX];

    if (argc == 2)
    {
        rt_strncpy(can_name, argv[1], RT_NAME_MAX);
    }
    else
    {
        rt_strncpy(can_name, CAN_DEV_NAME, RT_NAME_MAX);
    }
    /* 查找 CAN 设备 */
    can_dev = rt_device_find(can_name);
    if (!can_dev)
    {
        rt_kprintf("find %s failed!\n", can_name);
        return RT_ERROR;
    }
	
	
	
	

    /* 初始化 CAN 接收信号量 */
    rt_sem_init(&rx_sem, "rx_sem", 0, RT_IPC_FLAG_FIFO);

    /* 以中断接收及发送方式打开 CAN 设备 */
    res = rt_device_open(can_dev, RT_DEVICE_FLAG_INT_TX | RT_DEVICE_FLAG_INT_RX);
    RT_ASSERT(res == RT_EOK);
	
	
	/* 设置 CAN 通信的波特率为 500kbit/s*/
	res = rt_device_control(can_dev, RT_CAN_CMD_SET_BAUD, (void *)CAN500kBaud);
	RT_ASSERT(res == RT_EOK);
	
	/* 设置 CAN 的工作模式为正常工作模式 */
	res = rt_device_control(can_dev, RT_CAN_CMD_SET_MODE, (void *)RT_CAN_MODE_NORMAL);
	RT_ASSERT(res == RT_EOK);
	
	
    /* 创建数据接收线程 */
    thread = rt_thread_create("can_rx", can_rx_thread, RT_NULL, 1024, 25, 10);
    if (thread != RT_NULL)
    {
        rt_thread_startup(thread);
    }
    else
    {
        rt_kprintf("create can_rx thread failed!\n");
    }

    msg.id = 0x78;              /* ID 为 0x78 */
    msg.ide = RT_CAN_STDID;     /* 标准格式 */
    msg.rtr = RT_CAN_DTR;       /* 数据帧 */
    msg.len = 8;                /* 数据长度为 8 */
    /* 待发送的 8 字节数据 */
    msg.data[0] = 0x00;
    msg.data[1] = 0x11;
    msg.data[2] = 0x22;
    msg.data[3] = 0x33;
    msg.data[4] = 0x44;
    msg.data[5] = 0x55;
    msg.data[6] = 0x66;
    msg.data[7] = 0x77;
    /* 发送一帧 CAN 数据 */
    size = rt_device_write(can_dev, 0, &msg, sizeof(msg));
    if (size == 0)
    {
        rt_kprintf("can dev write data failed!\n");
    }

    return res;
}
/* 导出到 msh 命令列表中 */
MSH_CMD_EXPORT(can_sample, can device sample);
'''

wosayttn, 2021-12-08:
Hi,

Which CAN0 pins of M2354 are you used? Did you switch its pin multi-function to CAN function?
If not, you can use PinView tool to view all pins multi-function in MDK debugger. 

[image]

Leoxuyu, 2021-12-08:
I used NuTool_PinConfigure.exe to configure PA5 and PA4 as  multi-function to CAN function.Hardware Connection is OK.Now when I send a CAN frame,it can transmit success,but after transmit one CAN frame,it cant not go into  CAN0 interrupt entry function,so it stop work; 
[image]

Leoxuyu, 2021-12-08:
I put my code to the  github. You can try it!
https://github.com/Leoxuyu/M2354_RTT_CAN.git

wosayttn, 2022-01-05:
Sorry for later reply.
I solved this open-control order issue and made a PR on your repo.

[image]

mysterywolf, 2022-01-21:
@wosayttn 这个问题的pr提交到仓库了嘛

wosayttn, 2022-01-24:
Yes, please see #5501 .

mysterywolf, 2022-01-24:
OK

closed 2022-01-24 by mysterywolf