RT-Thread/rt-thread #5342: v4.1.0 discussion

Guozhanxin, 2021-12-07:
v4.0.4 已经发布了，所以我们开始筹备下个版本的发布了，下个版本按计划是一个次版本的更新，将会包含更多的新特性。

v4.0.4 has been released, so we have started to prepare for the release of the next version. The next version is planned to be a minor version update and will contain more new features.

**Features**

- [ ] MAL (mpu abstract layer) support.
- [ ] PSE51 support.

**BSP**

- [ ] Add STM32L5, STM32U5 BSP

**Package**

- [ ] LVGL
- [ ] TinyUSB

**Exquisite**

- [ ] Rename finsh to msh
- [ ] Update the file system elmfat version to R0.14b
- [ ] Remove RT_USING_TC https://github.com/RT-Thread/rt-thread/issues/1974
- [ ] Organize and clarify the level of DFS, libc and posix

**Others**

- [ ] Change the flag of mutex from FIFO to PRIO
- [ ] Optimize memory protection mechanism
- [ ] NewLib issue
- [ ] CFLAG flag issue

enkiller, 2021-12-07:
测试框架这部分有啥具体的动作吗？

Guozhanxin, 2021-12-08:
> 测试框架这部分有啥具体的动作吗？

直接移除掉吧

mysterywolf, 2021-12-09:
需要继续跟踪的PR
- [ ] #5296
- [ ] #5045

liukangcc, 2021-12-15:
https://github.com/RT-Thread/packages/tree/24f0a9c9d5d355fd88f416ca8e85c5ba05d52f91/language/pikascript

liukangcc, 2021-12-17:
https://github.com/RT-Thread/rt-thread/pull/5397

liukangcc, 2021-12-21:
MAL: https://github.com/RT-Thread/rt-thread/pull/5379

mysterywolf, 2021-12-24:
建议用__clang__代替__CLANG_ARM，__CLANG_ARM是我们在rtdef.h里自己定义的宏，使用受限，__clang__是标准的全局宏。
用#ifdef __ARMCC_VERSION代替#if defined(__CC_ARM) || defined(__CLANG_ARM)

mysterywolf, 2021-12-29:
https://club.rt-thread.org/ask/question/434195.html
ps或者list_thread最后一栏错误码，显示的是错误码代码 非常的不友好，别人根本不知道什么意思

mysterywolf, 2021-12-29:
需要对内核API的调用环境进行检查

mysterywolf, 2021-12-29:
LWIP2.0.2可以移除了 保留1.4.1 2.0.3 2.1.2三个版本即可

mysterywolf, 2021-12-31:
https://github.com/RT-Thread/rt-thread/blob/master/include/rtdef.h#L1063-L1166 需要从rtdef.h里移除掉的 这部分应该属于驱动框架.h负责的 应该是早起驱动框架没成熟的时候塞进去的

mysterywolf, 2022-01-03:
https://github.com/RT-Thread/rt-thread/blob/master/src/thread.c#L828-L832 此处操作非常的危险 ASSERT最好放在最开始的位置 不要放在某个分支里

ImFovery, 2022-01-05:
使用rtthread如鲠在喉的一个点：串口终端作为日志输出，即使ulog设置为异步也仍然为阻塞查询输出，占用cpu且输出效率低。

evan-liyb, 2022-01-07:
希望版本发布后，能与之前的所有包兼容。并且给出从老版本升级到新版本的官方文档。

mysterywolf, 2022-01-08:
> 前的所有包兼容。并且给出从老版本升级到新版本的官方文档。

https://github.com/RT-Thread/rt-thread/pull/5277
可以看一下这个，新版本有专门的legacy文件夹

evan-liyb, 2022-01-10:
打开之后，仍然很多不兼容的地方。特别是我自己写的应用层（大概8万行代码），用在新内核上很多地方不兼容了，主要是一些宏之类的做了调整，没去细看为什么要这样改，但希望尽量减少这种调整吧，或者用宏都老的定义管起来，但仍然保留。

mysterywolf, 2022-01-10:
> 打开之后，仍然很多不兼容的地方。特别是我自己写的应用层（大概8万行代码），用在新内核上很多地方不兼容了，主要是一些宏之类的做了调整，没去细看为什么要这样改，但希望尽量减少这种调整吧，或者用宏都老的定义管起来，但仍然保留。

再不改，里边很多东西都变成毒瘤了。

thewon86, 2022-01-20:
> 使用rtthread如鲠在喉的一个点：串口终端作为日志输出，即使ulog设置为异步也仍然为阻塞查询输出，占用cpu且输出效率低。

请看论坛我发的文章，我刚对串口框架做了个重构，阻塞 io 情况下，降低阻塞时间。

thewon86, 2022-01-20:
> https://club.rt-thread.org/ask/question/434195.html ps或者list_thread最后一栏错误码，显示的是错误码代码 非常的不友好，别人根本不知道什么意思

所有的 msh 命令，增加 c 函数调用接口，类似 https://gitee.com/thewon/rt_thread_repo/tree/master/user 这里实现的命令一样，既有终端命令入口，也有对等的 c 调用入口。

BernardXiong, 2022-01-24:
v4.1.0版本release后，可以拉分支出来。然后master也会尝试拉pre-v5.0.0分支，然后把smart相关的一些代码合并。如果春节期间能够合成功，那么再合并回master分支，做为后续的v5.0.x分支而存在。

另外，建议v4.1.0分支也不用做alpha, beta, rc版本，直接出v4.1.0版本，有问题可以快速修正到v4.1.1版本

BernardXiong, 2022-01-24:
> 打开之后，仍然很多不兼容的地方。特别是我自己写的应用层（大概8万行代码），用在新内核上很多地方不兼容了，主要是一些宏之类的做了调整，没去细看为什么要这样改，但希望尽量减少这种调整吧，或者用宏都老的定义管起来，但仍然保留。

可以一些点上多做沟通，如何在新特性和兼容性上做到平衡

mysterywolf, 2022-01-24:
> 另外，建议v4.1.0分支也不用做alpha, beta, rc版本，直接出v4.1.0版本，有问题可以快速修正到v4.1.1版本

建议用beta，因为4.1.0 beta版本还有一部分功能原计划4.1.0需要实现的还没有实现。

mysterywolf, 2022-01-24:
- [ ] 移除掉FINSH_FUNCTION_EXPORT FINSH_FUNCTION_EXPORT_ALIAS

mysterywolf, 2022-01-30:
https://github.com/RT-Thread/rt-thread/pull/5513#issuecomment-1025073518
pico的屏幕型号并没有指明

mysterywolf, 2022-01-31:
# Kernel

- [ ] Kernel中有使用int数据类型，应当予以替换掉
- [ ] struct thread结构体中采用隐式继承object，应改成显示继承 https://github.com/RT-Thread/rt-thread/pull/5500
- [ ] struct thread结构体需要清0 https://github.com/RT-Thread/rt-thread/pull/4891

mysterywolf, 2022-01-31:
## Scons
- [ ] 完善对cmake等脚本的支持 https://github.com/RT-Thread/rt-thread/issues/5482
- [ ] 补充Sconscript以及scons相关文档，由于RT-Thread采用的Scons作为构建工具本身就很不常见，再加上该构建工具是RTT自己魔改出来的，但是配套的文档并不全面，导致用户无法立即、使用Sconscript
- [ ] utest的头文件路径和group会被误加入到工程中

lliunix, 2022-01-31:
可否提供一个相对统一的 lcd 框架，对上提供统一的接口。这样在 lvgl 包中实现 rtt 的屏幕驱动，底层让用户和 bsp 去对接就好，然后直接编译就可以直接使用了。

mysterywolf, 2022-01-31:
> 可否提供一个相对统一的 lcd 框架，对上提供统一的接口。这样在 lvgl 包中实现 rtt 的屏幕驱动，底层让用户和 bsp 去对接就好，然后直接编译就可以直接使用了。

这个问题想过，当时我们自己维护的lvgl软件包（7.0）就是采用直接和rtt lcd框架对接的方式。但是经过实践之后发现，本身lvgl自己就已经提供了底层的抽象层，如果再对接一个lcd框架就显得脱裤子放屁，因此在lvgl官方软件包中，就采用了lvgl原生的api，方便用户做出选择，既可以直接怼lcd原生驱动，也可以自行怼lcd框架。我们这侧就不强制用户非得使用lcd的框架了。

mysterywolf, 2022-02-08:
## exmaple
- [ ] 移除部分文件夹，转移到sample软件包
- [ ] 在example目录下增加project文件夹，作为项目实验室，为多BSP提供完成的项目级例程体验（还需多讨论，dist时是复制过去 or 参照utest？）
- [ ] utest的头文件路径和group会被误加入到工程中，UTEST_SELF_PASS_TC应该default为n
- [ ]  在example目录下增加project文件夹，作为项目实验室，为多BSP提供完成的项目级例程体验（还需多讨论，dist时是复制过去 or 参照utest？）

mysterywolf, 2022-02-08:
## LWIP
- [ ] https://github.com/RT-Thread/rt-thread/pull/5587 
- [ ] 推动upstream或与upstream保持同步

mysterywolf, 2022-02-08:
## LVGL

- [ ] https://github.com/RT-Thread/rt-thread/pull/5513#issuecomment-1025073518 pico的屏幕型号并没有指明
- [ ] stm32f469 驱动越过驱动框架，驱动框架里又含着两个buffer，这样依赖相当于驱动用了4-5个屏幕大小的buffer

thewon86, 2022-02-09:
https://github.com/RT-Thread/rt-thread/pull/4740#issuecomment-1033219023

mysterywolf, 2022-02-11:
## 需要向pre-5.0.0同步的pr
- [ ] https://github.com/RT-Thread/rt-thread/pull/5589
- [ ]

mysterywolf, 2022-02-17:
https://club.rt-thread.org/ask/article/3392.html

jifanchn, 2022-03-02:
建议考虑CMake作为未来的构建工具，现在越来越多的IDE都在支持CMake了，scons确实有点老了。

目前正在考虑把公司的下一代产品构建在RTThread上，但是这个scons的构建系统让人望而却步。

BernardXiong, 2022-03-02:
> 建议考虑CMake作为未来的构建工具，现在越来越多的IDE都在支持CMake了，scons确实有点老了。
> 
> 目前正在考虑把公司的下一代产品构建在RTThread上，但是这个scons的构建系统让人望而却步。

可以提一份方案上来，cmake也有些古老，其中的语法方式也不那么友好。scons部分在RTT这边做了挺多的一些调整，有好的有坏的，一下就想更换构建系统，不容易/很不容易。

mysterywolf, 2022-03-02:
@BernardXiong @jifanchn 坚决反对将RT-Thread Scons转移到其他构建方式。这相当于是自废武功。scons构建方式很灵活，基于python语法的优势，很多复杂的文件和路径添加几句话就可以完成，而且可以引用python丰富的内置函数。Scons是RT-Thread定制化的构建工具，我们可以根据需求自行对scons进行修改，处理更加灵活。
此外RT-Thread支持Keil，IAR VSCODE VS Cmkae Makefile等其他工程的生成，这也全都得益于scons。如果转换到Cmake构建方式，如何构建Keil IAR等等其他的project？Scons是支持生成Cmake工程的，在env里，敲`scons --target=cmake` 即可
目前Scons的主要问题是文档太少了！这也是很多人望而却步的主要原因，实际上常用的Scons函数没有几个，如果有文档或者教学视频，照葫芦画瓢很简单。文档和教学视频是RT-Thread目前依然要急需填的坑！

jifanchn, 2022-03-03:
@mysterywolf 如果能用SCONS完美构建CMAKE工程也是可以接受的，但是也要考虑到公司的静态库权限管控和SDK升级需求。Keil和IAR这样的IDE我们已经深受其害了，在多人维护的系统和持续迭代更新的产品中，编译目标的控制无法被GIT等源代码版本控制系统维护是造成嵌入式产品源代码碎片化的重要原因之一。另外构建系统不仅需要支持嵌入式代码编译，也要能支持整个软件生命周期的源代码测试用例、并行仿真、第三方算法工程代码生成、开源代码的引用的调用等一系列操作。CMAKE的开发者数量与生态已经不是SCONS能比的了，甚至可以说已经是C/C++事实上的标准构建系统了。想想QMAKE/MAKEFILE，甚至微软VS的构建系统都已经转向支持CMAKE，JB家的CLION甚至只支持CMAKE。换句话说，SCONS生成Keil和IAR这样的工程，也只是对不熟悉开源社区的“传统”嵌入式程序员的兼容罢了。看看还活跃的C/C++开源库，不支持CMake的太少了。缺少说明文档库，其实就是生态缺失的重要标志之一。
RTThread作为国产的优秀、稳定的嵌入式操作系统，不仅应该做到兼容，同时也应考虑能“温和地”正确引领嵌入式开发的发展方向，为嵌入式开发生态转变甚至超越目前的互联网生态模式做出更多努力。
这也是我建议转向CMake的原因之一。

----看看zephyr吧。

mysterywolf, 2022-03-03:
zephyr的模式学不来的，门槛太高了，得要求使用者有Linux使用经验，全面转到Linux环境或类unix环境下开发，其开发方式很不符合我国国情。

中国开发嵌入式的工程师来自于两部分，一部分是计算机专业以及软件工程专业毕业的，受过Linux等良好的软件开发教育，可以轻松应对 Cmake，makefile，shell等等这些东西。还有一部分是电子、电子、通信专业毕业，其专业内，Linux等软件开发科目并不是必修科目，他们更喜欢用Keil IAR这类集成性IDE来点点点按钮。

当然如果你觉得目前RTT CMake应该如何做，欢迎提出意见，因为我们确实也收到类似希望采用CMake来开发的建议了。但是很难全面转型到Cmake，就像上面说的，我们会完善scons的Cmake脚本生成能力，或者下次发布版本的时候可以把Cmake脚本直接刷到仓库里，让CMake用户可以直接用起来，直接无视Scons，这都是可以讨论的。但是scons肯定是不能动的，我们得保证中国所有专业方向的嵌入式工程师的死活，不能仅仅服务于那一部分非常专业的、熟悉Linux开发环境的嵌入式工程师。

jifanchn, 2022-03-03:
同意你的说法，keil和IAR是要支持的。

CMake能有某种方法直接和scons并行这是最好的。

互联网最近已经开始卷了，国内未来深入设备底层的开发也会越来越多，我司目前也看到招聘中人员收入，学历和行业的错位。未来也会有越来越多的熟悉Linux的专业程序员进入。

mysterywolf, 2022-03-03:
# CMake
- [ ] 完善通过Scons生成CMake的能力
目前，Scons生成的CMake脚本只是一个文件放在bsp的根目录下，应参考如micropython等其它CMake构建的开源项目的CMakeLists.txt分配模式，可能每个文件夹都有一个CMakeLists.txt，多级，而不是全塞在一个CMakeLists.txt中。
- 上述功能实现的情况下，可以考虑下次版本发布时，直接将CMakeLists.txt刷到各个文件夹下，让熟悉CMake的用户直接用CMake，越过Scons，考虑这种可能行。

aozima, 2022-03-04:
- [ ] 支持64位时间戳，以解决千年虫问题
[image]

mysterywolf, 2022-03-04:
- [ ] STM32 LPUART的问题
  - https://club.rt-thread.org/ask/question/424812.html
  - https://club.rt-thread.org/ask/article/3412.html

mysterywolf, 2022-03-04:
# 软件包
在4.1.0需要升级的：
- [ ] https://github.com/RT-Thread-packages/paho-mqtt
- [ ] https://github.com/RT-Thread-packages/fal FAL根据龙哥建议，应当放到主仓库

BreederBai, 2022-03-08:
新版本有计划增加获取USB是否连接的接口吗？USB DTR标志是必须的吗？ChibiOS的usb驱动（STM32F427），上位机勾选DTR，似乎也能通信

cha331, 2022-03-08:
> # 软件包
> 在4.1.0需要升级的：
> 
> * [ ]  https://github.com/RT-Thread-packages/paho-mqtt issue中反馈了一些bug并提出了解决方案，需要整合到仓库中
> * [ ]  https://github.com/RT-Thread-packages/fal FAL根据龙哥建议，应当放到主仓库
> * [ ]  https://club.rt-thread.org/ask/question/433049.html
> * [ ]  https://club.rt-thread.org/ask/question/431576.html
> * [ ]  https://club.rt-thread.org/ask/question/430162.html

libmodbus软件包编译报错问题也可以修改下，主要涉及几个头文件，论坛里已有解决方案
https://club.rt-thread.org/ask/article/3280.html

mysterywolf, 2022-03-09:
STM32F1 RTC读写日期不正常 https://club.rt-thread.org/ask/question/434910.html

mysterywolf, 2022-03-09:
## deinit 低功耗特性
- [ ] https://github.com/RT-Thread/rt-thread/issues/611
设备close的时候只是关闭了设备，但是设备寄存器配置的内容并没有被清空。
1. 无法进入低功耗状态
2. 例如某引脚作为IO输出，再进入其他模式（如pwm）就会出现问题。

tfx2001, 2022-03-11:
> 新版本有计划增加获取USB是否连接的接口吗？USB DTR标志是必须的吗？ChibiOS的usb驱动（STM32F427），上位机勾选DTR，似乎也能通信

@BreederBai 是 RTT 的 USB 协议栈还是 TinyUSB 呢？

liukangcc, 2022-03-11:
https://github.com/RT-Thread-packages/paho-mqtt/pull/47

liukangcc, 2022-03-11:
FAL添加到 rt-thread 主仓库：https://github.com/RT-Thread/rt-thread/pull/5662

mysterywolf, 2022-03-11:
[image]
在gcc下 uint32_t是long unsigned int ，而在keil下是 unsigned int 。这个不同编译器定义不一样。
rt_uint32_t是 unsigned int，导致rt_uint32_t的函数形参在填入uint32_t的变量时会出现警告。

mysterywolf, 2022-03-11:
## Finsh 疑似bug
- [ ] https://club.rt-thread.org/ask/question/434691.html
- [ ] https://club.rt-thread.org/ask/question/430704.html

mysterywolf, 2022-03-12:
[image]
内核可以考虑用这种方式来设置堆栈大小，例如对于龙芯，其idle main线程的栈大小要高于32位的

liukangcc, 2022-03-15:
> ## Finsh 疑似bug
> * [ ]  https://club.rt-thread.org/ask/question/434691.html
> * [ ]  https://club.rt-thread.org/ask/question/430704.html

第二个源文件被删了吧 https://club.rt-thread.org/ask/question/430704.html

mysterywolf, 2022-03-15:
> 第二个源文件被删了吧 https://club.rt-thread.org/ask/question/430704.html

是的可以不用管了~

BreederBai, 2022-03-15:
> > 新版本有计划增加获取USB是否连接的接口吗？USB DTR标志是必须的吗？ChibiOS的usb驱动（STM32F427），上位机勾选DTR，似乎也能通信
> 
> @BreederBai 是 RTT 的 USB 协议栈还是 TinyUSB 呢？

@tfx2001  RTT 的 USB 协议栈

liukangcc, 2022-03-16:
> # 软件包
> 在4.1.0需要升级的：
> 
> * [x]  https://github.com/RT-Thread-packages/paho-mqtt issue中反馈了一些bug并提出了解决方案，需要整合到仓库中
> * [x]  https://github.com/RT-Thread-packages/fal FAL根据龙哥建议，应当放到主仓库
> * [ ]  https://club.rt-thread.org/ask/question/433049.html
> * [ ]  https://club.rt-thread.org/ask/question/431576.html
> * [x]  https://club.rt-thread.org/ask/question/430162.html
> * [x]  https://club.rt-thread.org/ask/question/430162.html
> * [ ]  libmodbus软件包编译报错问题也可以修改下，主要涉及几个头文件，论坛里已有解决方案
>   https://club.rt-thread.org/ask/article/3280.html

第三个作者修了 https://github.com/gbcwbz/canfestival-rtt/issues/9#issuecomment-1068134572

mysterywolf, 2022-03-21:
TODO:
https://github.com/RT-Thread/rt-thread/pull/5685
https://github.com/RT-Thread/rt-thread/pull/5661
https://github.com/RT-Thread/rt-thread/pull/5660

mysterywolf, 2022-03-22:
尽量压缩C99的使用需求，如无特殊需求使用CFLAGS或者LOCAL_CFLAGS, 而非CCFLAGS，不要讲C++和C的配置项默认混在一起。
部分编译器如（Keil-MDK），针对C++和C的配置要求不一样，混在一起配置可能会出问题。

[7 more replies]

closed 2022-06-21 by mysterywolf