RT-Thread/rt-thread #5346: 野火霸道V2F103板卡ESP8266适配不完整！

hessonsu, 2021-12-08:
cd到这个目录
bsp\stm32\stm32f103-fire-arbitrary
[启用ESP8266](https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-standard/application-note/components/at/an0014-at-client)
调试之后发现wait AT client connect timeout
经过排查。发现板卡级别设配无初始化引脚PIN_G13,PIN_G14.这2个引脚。
原理图如下。
[image]

待修复代码为：

#define WIFI_EN_PIN         GET_PIN(G, 13)
#define WIFI_RESET_PIN      GET_PIN(G, 14)

    rt_pin_mode(WIFI_EN_PIN, PIN_MODE_OUTPUT);
    rt_pin_mode(WIFI_RESET_PIN, PIN_MODE_OUTPUT);
    
    rt_pin_write(WIFI_RESET_PIN, PIN_LOW);
    rt_thread_mdelay(500);
    rt_pin_write(WIFI_RESET_PIN, PIN_HIGH);
    
    rt_pin_write(WIFI_EN_PIN, PIN_HIGH);

hessonsu, 2021-12-08:
我还在学习阶段。主要是不知道这个代码加在哪里比较适合。不然我就提一个合并请求了。提上来。也是希望有人提交一下给我学习

Guozhanxin, 2021-12-08:
棒👍

mysterywolf, 2021-12-09:
https://www.bilibili.com/video/BV1gr4y1w7yX 参见这个教程

hessonsu, 2021-12-15:
野火霸道STM32F103ZET6  扩展SRAM

 https://blog.csdn.net/zcy_cyril/article/details/106521319
按上面的连接配置CubeIDE 并重新生成代码

https://github.com/RT-Thread/rt-thread/tree/ef71de34f76d4403fcd2cddd21a17f3a53a40296/bsp/stm32/stm32f407-atk-explorer/board/ports
到这个地方拷贝drv_sram.c drv_sram.h这2个文件

修改drv_sram.c 代码。使他适配宏名义
修改#ifdef BSP_USING_SRAM 为 #ifdef BSP_USING_EXT_SRAM

修改board目录下的Kconfig文件
新增代码
menuconfig BSP_USING_EXT_SRAM
        bool "Enable external sram"
        select RT_USING_MEMHEAP
        select RT_USING_MEMHEAP_AS_HEAP
        default n
        if BSP_USING_EXT_SRAM
            config BSP_EXT_SRAM_SIZE
            hex "external sram size"
            default 0x100000
        endif
该代码放在menu "Onboard Peripheral Drivers"这一级下面

再修改宏定义为#define RT_USING_MEMHEAP_AS_HEAP