RT-Thread/rt-thread #5363: Rt-thread on Teensy 4.1 , Please

JPB1JPB1, 2021-12-10:
Rt-thread on Teensy 4.1 ,4.x , please
https://www.pjrc.com/store/teensy41.html

BSP, 4 bit Sdio-sd-card... USB host for wifi, bluetooth, low power mode ...

Any plans , 
Best Regards

hinata9276, 2022-03-17:
Yes, I believe it will work. Teensy 4.0 and 4.1 use MIMXRT1062 MCU, highly compatibility with the officially supported NXP MIMXRT10624 evaluation kit. I can't imagine why it does not support Teeny 4.

Guozhanxin, 2022-03-28:
Maybe you can make a bsp yourself. Here is a complete set of bsp production tutorials. Follow the tutorial, very easy to support Teensy 4 

https://github.com/RT-Thread/rt-thread/blob/master/bsp/imxrt/docs/IMXRT%E7%B3%BB%E5%88%97BSP%E5%88%B6%E4%BD%9C%E6%95%99%E7%A8%8B.md

closed 2024-12-15 by Rbb666