RT-Thread/rt-thread #5369: RA6m4-CPK BSP无法编译通过

evan-liyb, 2021-12-13:
RA6M4 CPK的BSP无法编译通过，KEIL打印如下。应该是工程生成有什么问题，或者有没有上传的文件？
[image]

mysterywolf, 2021-12-13:
你用menuconfig刷新一下keil工程？

evan-liyb, 2021-12-13:
我执行了menconfig,然后直接保存，再执行scons --target=mdk，生成新的工作后。还是不行，现象一样。 
莫非是对keil版本有什么特殊需求？又或者并非像readme说的那样，直接可以打开工程编译么

evan-liyb, 2021-12-13:
我执行了menuconfig,然后直接保存，再执行scons --target=mdk，生成新的工程后。还是不行，现象一样。
莫非是对keil版本有什么特殊需求？又或者并非像readme说的那样，直接可以打开工程编译么?

Guozhanxin, 2021-12-13:
还有一个pr在处理，预计今天就会合并了，你等 https://github.com/RT-Thread/rt-thread/pull/5262 这个pr合并之后再试。

evan-liyb, 2021-12-13:
好的，感谢答复。
待我把RT-thread学精通了，我也会贡献代码。

evan-liyb, 2021-12-14:
各位RT开发大家好，麻烦务必找一台新环境测试一下是否新的RA64的BSP代码能正常编译，这里面的很多编译问题涉及scon构建的原理，我们自己解决起来很难。

mysterywolf, 2021-12-14:
啥意思 什么叫新的环境？没懂

mysterywolf, 2021-12-14:
最新的pr已经合并了 你可以下载一份最新的代码测试一下

evan-liyb, 2021-12-14:
Hello, 我更新新代码，看到了相关更新日志，但是看上去好像还是不能编译。我怀疑是不是我的使用方法的问题，确证一下，在我只需要把rtt在该开发板跑起来的情况下，只需要下载一份完整的代码然后直接打开工程编译就可以？是否对keil的版本有什么特定要求呢?我的KEIL版本是V5.24.2

evan-liyb, 2021-12-14:
新环境是指：找台非开发人员使用的PC，拉取一份新代码，尝试编译。
不好意思，之前确实没描述清楚。

mysterywolf, 2021-12-14:
理论上，下载一份最新的代码，然后进到menuconfig什么也不干退出去来刷一下工程，因为这个是开发分支所以你需要手动刷一下工程，scons --target=mdk5 就可以了 然后打开keil工程就可以用了。

mysterywolf, 2021-12-14:
注意是scons --target=mdk5 不是mdk，mdk貌似是给keil2刷工程

evan-liyb, 2021-12-14:
我刚刚重新把目录还原，然后menuconfig没修改直接保存，用scons --target=mdk5重新生成了工程发现现象是一样的。
方便通过远程桌面工具远程支持一下吗，或者对于你们来说十分钟就解决了，但对我来说可能需要很久（因为对这套机制还不熟悉）。我们公司叫映翰通网络，也是你们的官网上能查到到合作伙伴

mysterywolf, 2021-12-14:
额 我是RT-Thread社区的不是睿赛德的。不过我注意到一个问题，你用的是ARM AC6不是AC5 你试试用AC5？

evan-liyb, 2021-12-14:
额，你有跑通过这个BSP吗，我死活不行。
换成AC5直接说不支持M33
[image]

Guozhanxin, 2021-12-14:
@ShermanShao

mysterywolf, 2021-12-14:
@ShermanShao  他刚提交完这个bsp的PR，等他回复吧

ShermanShao, 2021-12-14:
或许是MDK版本和工具链版本导致的。下图是测试过的环境版本
[image]

[image]

ShermanShao, 2021-12-14:
@evan-liyb   找到一篇文档或许可以解释这个问题。 MDK 不认识 SConscript 这个文件，但是在我的环境下是没出现这个报错的。
https://www.keil.com/support/man/docs/uv4/uv4_ca_usecustomtranslator.htm
[image]

evan-liyb, 2021-12-14:
我正在下载最新的版本，然后安装来调试，我的MDK工具链版本确实比你的低不少。不管怎么样，更新IDE都是要做的。
[image]

ShermanShao, 2021-12-14:
@evan-liyb 或许可以尝试下这个方式，让 MDK 把 SConscript 识别成文档类型，SConscript 文件在MDK 中是无用的，是因为 RASC 工具会把 src 目录下的文件都加入到工程中，所以没法手动移除。如果确认是这个问题的话，可以尽快修复掉。 
麻烦帮忙验证下是不是可行
https://www.keil.com/support/man/docs/uv4/uv4_ca_file_extensions.htm

[image]

ShermanShao, 2021-12-14:
@evan-liyb  可以先试试看上面的方法，如果可行我就修复一下

evan-liyb, 2021-12-14:
@ShermanShao 我仅仅从官网上下载了MDK5.36版本，然后再编译，看上去有改变，首先它识别到了SConscript文件，其次是在编译工程里的文件了（之前是直接没有编译工程里的文件）。只是有报错，缺少一个.h文件。我是否需要手动用RASC生成一下代码呢
我现在的KEIL版本如下
[image]

编译日志如下：
Rebuild started: Project: project
*** Using Compiler 'V6.16', folder: 'D:\Keil_v5\ARM\ARMCLANG\Bin'
Rebuild target 'Target 1'
compiling div0.c...
compiling backtrace.c...
compiling showmem.c...
assembling context_rvds.S...
compiling cpuport.c...
compiling pin.c...
compiling serial_v2.c...
compiling completion.c...
compiling waitqueue.c...
compiling ringbuffer.c...
compiling pipe.c...
compiling dataqueue.c...
compiling workqueue.c...
compiling ringblk_buf.c...
drivers/drv_gpio.c(11): warning: In file included from...
drivers\drv_gpio.h(20): warning: In file included from...
./ra_gen\hal_data.h(7): warning: In file included from...
./ra/fsp/inc/instances\r_icu.h(33): error: 'r_external_irq_api.h' file not found
#include "r_external_irq_api.h"
         ^~~~~~~~~~~~~~~~~~~~~~
1 error generated.
compiling drv_gpio.c...
drivers/drv_usart_v2.c(11): warning: In file included from...
drivers\drv_usart_v2.h(18): warning: In file included from...
drivers/config\drv_config.h(22): warning: In file included from...
drivers/config/ra6m4/uart_config.h(15): warning: In file included from...
./ra_gen\hal_data.h(7): warning: In file included from...
./ra/fsp/inc/instances\r_icu.h(33): error: 'r_external_irq_api.h' file not found
#include "r_external_irq_api.h"
         ^~~~~~~~~~~~~~~~~~~~~~
1 error generated.
compiling drv_usart_v2.c...
drivers/drv_common.c(16): warning: In file included from...
drivers\drv_gpio.h(20): warning: In file included from...
./ra_gen\hal_data.h(7): warning: In file included from...
./ra/fsp/inc/instances\r_icu.h(33): error: 'r_external_irq_api.h' file not found
#include "r_external_irq_api.h"
         ^~~~~~~~~~~~~~~~~~~~~~
1 error generated.
compiling drv_common.c...
compiling shell.c...
compiling msh.c...
compiling cmd.c...
compiling components.c...
compiling kservice.c...
compiling mem.c...
compiling clock.c...
compiling idle.c...
compiling thread.c...
compiling ipc.c...
compiling scheduler.c...
compiling device.c...
compiling irq.c...
compiling timer.c...
compiling object.c...
compiling time.c...
src/hal_entry.c(13): warning: In file included from...
./ra_gen\hal_data.h(7): warning: In file included from...
./ra/fsp/inc/instances\r_icu.h(33): error: 'r_external_irq_api.h' file not found
#include "r_external_irq_api.h"
         ^~~~~~~~~~~~~~~~~~~~~~
1 error generated.
compiling hal_entry.c...
compiling board_init.c...
compiling board_leds.c...
compiling startup.c...
J:/rt-thread/bsp/ra6m4-cpk/ra/fsp/src/bsp/cmsis/Device/RENESAS/Source/system.c(31): warning: In file included from...
./ra_gen\hal_data.h(7): warning: In file included from...
./ra/fsp/inc/instances\r_icu.h(33): error: 'r_external_irq_api.h' file not found
#include "r_external_irq_api.h"
         ^~~~~~~~~~~~~~~~~~~~~~
1 error generated.
compiling system.c...
compiling bsp_clocks.c...
compiling bsp_common.c...
compiling bsp_delay.c...
compiling bsp_group_irq.c...
compiling bsp_guard.c...
compiling bsp_io.c...
compiling bsp_irq.c...
compiling bsp_register_protection.c...
compiling bsp_rom_registers.c...
compiling bsp_sbrk.c...
compiling bsp_security.c...
J:/rt-thread/bsp/ra6m4-cpk/ra/fsp/src/r_icu/r_icu.c(24): warning: In file included from...
./ra/fsp/inc/instances\r_icu.h(33): error: 'r_external_irq_api.h' file not found
#include "r_external_irq_api.h"
         ^~~~~~~~~~~~~~~~~~~~~~
1 error generated.
compiling r_icu.c...
compiling r_ioport.c...
compiling common_data.c...
compiling r_sci_uart.c...
J:/rt-thread/bsp/ra6m4-cpk/ra_gen/hal_data.c(2): warning: In file included from...
./ra_gen/hal_data.h(7): warning: In file included from...
./ra/fsp/inc/instances\r_icu.h(33): error: 'r_external_irq_api.h' file not found
#include "r_external_irq_api.h"
         ^~~~~~~~~~~~~~~~~~~~~~
1 error generated.
compiling hal_data.c...
J:/rt-thread/bsp/ra6m4-cpk/ra_gen/main.c(2): warning: In file included from...
./ra_gen/hal_data.h(7): warning: In file included from...
./ra/fsp/inc/instances\r_icu.h(33): error: 'r_external_irq_api.h' file not found
#include "r_external_irq_api.h"
         ^~~~~~~~~~~~~~~~~~~~~~
1 error generated.
compiling main.c...
compiling pin_data.c...
compiling vector_data.c...
".\Objects\ra6m4.axf" - 8 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:00:07

ShermanShao, 2021-12-14:
这个文件应该是我漏掉了，我一会更新下。 如果已经装了 RASC 可以生成下试试，应该就可以了

evan-liyb, 2021-12-14:
我先用RASC生成了代码，然后把git管起来的代码还原。编译通过了。

感谢各位。
 @ShermanShao 
 @mysterywolf

mysterywolf, 2021-12-14:
https://github.com/RT-Thread/rt-thread/pull/5380/files

mysterywolf, 2021-12-14:
如果没有问题了的话 可以关闭此issue 如果有问题可以继续提问

evan-liyb, 2021-12-15:
已关闭。

closed 2021-12-15 by evan-liyb