RT-Thread/rt-thread #5378: gd32e230k-start bsp无法编译通过

pikasTech, 2021-12-14:
在使用gd32e230k-start bsp时，遇到编译错误：

```
ArmClang: error: unsupported option '--c99'
```

以下是完整的log

```
Rebuild started: Project: project
*** Using Compiler 'V6.16', folder: 'C:\Keil_v5\ARM\ARMCLANG\Bin'
Rebuild target 'rt-thread_gd32e230'
assembling context_rvds.S...
compiling cpuport.c...
ArmClang: error: unsupported option '--c99'
compiling i2c_dev.c...
ArmClang: error: unsupported option '--c99'
compiling serial.c...
ArmClang: error: unsupported option '--c99'
compiling sfud.c...
ArmClang: error: unsupported option '--c99'
compiling i2c_core.c...
compiling div0.c...
compiling showmem.c...
ArmClang: error: unsupported option '--c99'
compiling pin.c...
compiling main.c...
compiling backtrace.c...
ArmClang: error: unsupported option '--c99'
compiling spi_core.c...
ArmClang: error: unsupported option '--c99'
compiling spi_flash_sfud.c...
ArmClang: error: unsupported option '--c99'
compiling spi_dev.c...
ArmClang: error: unsupported option '--c99'
compiling ringblk_buf.c...
ArmClang: error: unsupported option '--c99'
compiling waitqueue.c...
ArmClang: error: unsupported option '--c99'
compiling ringbuffer.c...
ArmClang: error: unsupported option '--c99'
compiling workqueue.c...
ArmClang: error: unsupported option '--c99'
compiling dataqueue.c...
ArmClang: error: unsupported option '--c99'
compiling pipe.c...
ArmClang: error: unsupported option '--c99'
compiling completion.c...
compiling drv_i2c.c...
compiling board.c...
compiling drv_gpio.c...
compiling drv_usart.c...
compiling drv_spi.c...
compiling msh.c...
compiling scheduler.c...
compiling shell.c...
compiling cmd.c...
compiling clock.c...
compiling mem.c...
compiling thread.c...
compiling mempool.c...
compiling kservice.c...
compiling device.c...
compiling object.c...
compiling irq.c...
compiling components.c...
compiling timer.c...
compiling idle.c...
compiling time.c...
compiling ipc.c...
compiling gd32e230_cmp.c...
compiling gd32e230_exti.c...
compiling gd32e230_wwdgt.c...
compiling gd32e230_syscfg.c...
compiling gd32e230_adc.c...
assembling startup_gd32e230.s...
compiling gd32e230_timer.c...
compiling gd32e230_gpio.c...
compiling gd32e230_i2c.c...
compiling gd32e230_spi.c...
compiling system_gd32e230.c...
compiling gd32e230_dbg.c...
compiling gd32e230_fwdgt.c...
compiling gd32e230_fmc.c...
compiling gd32e230_dma.c...
compiling gd32e230_rcu.c...
compiling gd32e230_misc.c...
compiling gd32e230_pmu.c...
compiling gd32e230_crc.c...
compiling gd32e230_rtc.c...
compiling gd32e230_usart.c...
".\build\rtthread-gd32e230.axf" - 15 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:00:04
```

mysterywolf, 2021-12-15:
这个是因为你用的是AC6  这个bsp没有区分AC5和AC6  你暂时先用AC5吧

pikasTech, 2021-12-17:
这个工程没有ac5的选项，readme也是写的ac6 （捂脸）

pikasTech, 2021-12-17:
[image]~~~~

mysterywolf, 2021-12-17:
很尴尬

Guozhanxin, 2021-12-21:
修

Guozhanxin, 2021-12-21:
https://github.com/RT-Thread/rt-thread/pull/5409

closed 2021-12-22 by Guozhanxin