RT-Thread/rt-thread #5398: main.c 里面访问https接口报错

hessonsu, 2021-12-17:
为啥在main.c 里面访问https接口会出现问题。
assertion failed at function:rt_thread_suspend, line numbe

试过用webclient 的例子在msh 调用就无问题
web_get_test   https://www.rt-thread.com/service/rt-thread.txt

xiangxistu, 2021-12-17:
你这个错误是断言错误，原因很多，最有可能就是栈溢出问题。

msh 与 main 的区别；

1. main 函数，实际上是一个 main 线程；默认栈大小为 2048 ，优先级为 10；main 线程的启动，是 RT-Thread 系统初始化完成后，创建的一个线程，其中的工作包含了一些自动初始化工作，最后是你写的 main 函数中的逻辑；如果此时有些设备还没有完成初始化，哪怕没有出现刚才的断言错误，也有可能出现一些逻辑错误。
2. msh 执行命令，实际上是使用 tshell 线程，默认栈大小为 4096，优先级为 20；

hessonsu, 2021-12-17:
@xiangxistu 果然是栈太小了。我设置4096就可以了。

我在main 线程中，也有等待网络连接可用状态。
struct netdev * dev1= netdev_get_by_family(AF_AT);
  while (1)
  {
    
    if (netdev_is_internet_up(dev1) > 0)//判断ESP8266是否联网成功
    {
         https接口调用
    }
    //rt_thread_mdelay(1000);
}

closed 2021-12-17 by hessonsu