RT-Thread/rt-thread #5461: Since commit #5437 I get a compiler error in syscalls.c

ThomasHornschuh, 2021-12-31:
Since commit #5437 when compiling my project I get the error
```
CC build/kernel/components/libc/compilers/gcc/newlib/syscalls.o
/home/thomas/development/rt-thread/components/libc/compilers/gcc/newlib/syscalls.c: In function '_write_r':
/home/thomas/development/rt-thread/components/libc/compilers/gcc/newlib/syscalls.c:303:9: error: unknown type name 'rt_device_t'; did you mean 'rt_dev_t'?
````
I'm currently porting RT-Thread to my own FPGA based RISC-V Softcore, so I did not try to reproduce the problem with a upstream BSP. Because I'm at the beginning of my port I have not implemented any device driver, so disabled "using device object" in "Kernel Device Object" config. My full config can be found in https://github.com/bonfireprocessor/rt-thread/blob/master/bsp/bonfire_arty/.config

Maybe my configuration is deprecated, but disabling the kernel device object seems to be needed that finsh is using the `rt_hw_console_getchar` function for input. 
As workaround I patched syscalls.c to contain the RT_USING_LIBC macro again A better solution would be to change the code of _write_r to to check RT_USING_DEVICE

mysterywolf, 2021-12-31:
>Maybe my configuration is deprecated

This is my problem. 
Thank you for you report, I will add `RT_USING_DEVICE`.

mysterywolf, 2021-12-31:
https://github.com/RT-Thread/rt-thread/pull/5462/files

mysterywolf, 2022-01-08:
Problem has been solved, thank you! https://github.com/RT-Thread/rt-thread/pull/5491

closed 2022-01-08 by mysterywolf