RT-Thread/rt-thread #5466: [bsp] tm4c129xl eth_drv fault;

rqbh, 2022-01-02:
arm-none-eabi-gcc:    v9.2.1
git branch : master
cross compiler: gcc

只在iot软件中心中启用freemodbus slave tcp, 其他保持默认
编译后烧写到ek_tm4c129xl 板子中，运行

不插网线没问题
插入网线出现如下fault

r00: 0x20000ce4
r01: 0x00000000
r02: 0x20000ce4
r03: 0x20008ef4
r04: 0xdeadbeef
r05: 0xdeadbeef
r06: 0xdeadbeef
r07: 0x20009358
r08: 0xdeadbeef
r09: 0xdeadbeef
r10: 0xdeadbeef
r11: 0xdeadbeef
r12: 0x00000000
 lr: 0x00021297
 pc: 0x20008ef4
hard fault on thread: erx

thread   pri  status      sp     stack size max used left tick  error
-------- ---  ------- ---------- ----------  ------  ---------- ---
tshell    20  suspend 0x00000180 0x00001000    09%   0x0000000a 000
etx       12  suspend 0x000000ac 0x00000400    16%   0x00000010 000
erx       12  running 0x0000013c 0x00000400    37%   0x00000010 000
tcpip     10  suspend 0x000000f4 0x00000400    51%   0x00000013 000
sys_work  23  suspend 0x00000084 0x00000800    06%   0x0000000a 000
tidle0    31  ready   0x00000060 0x00000100    59%   0x00000002 000
timer      4  suspend 0x00000080 0x00000200    25%   0x0000000a 000
usage fault:
SCB_CFSR_UFSR:0x02 INVSTATE

rqbh, 2022-01-02:
@ArdaFu 我看这个bsp是你维护的，默认你用的keil，请问使用gcc有没有测试过 ？
在ubuntu 21.10 x64 平台

rqbh, 2022-01-03:
用KEIL编译下载到板子后，上电也无法dhcp得到ip

ArdaFu, 2022-01-03:
这个应该好久没有更新过了 我在出差现在没有这个板子在手上 无法测试。

ArdaFu, 2022-01-03:
另外我当时应该是用keil编译的，没有在linux的gcc试过。

ArdaFu, 2022-01-03:
你可以debug到你的pc: 0x20008ef4这个地址看看是哪段代码，hard fault 提示是 接收线程失败了。

closed 2022-02-17 by mysterywolf