RT-Thread/rt-thread #5476: 树莓派移植报错

xiaoyu4122, 2022-01-05:
[image]

xiaoyu4122, 2022-01-05:
在编译系统的时候，输入scons，会报这样的错。但是结构体里面有这个成员

mysterywolf, 2022-01-05:
related: https://github.com/RT-Thread/rt-thread/pull/5336
@chenyingchun0312 迎春 看一下这个问题呗，应该是bsp也应该升级一下

chenyingchun0312, 2022-01-05:
我去看了一下代码：在raspberry-pi\raspi3-64\driver\drv_i2c.c 和raspberry-pi\raspi3-32\driver\drv_i2c.c 这两个BSP的I2C Driver 中，使用到了bus->addr， 但实际上这样使用，应该是不合理的，在函数`raspi_i2c_mst_xfer`中，获取具体的I2C控制器实例，怎么能通过bus->addr是否为0来判断呢？ 我的理解，在注册I2C总线设备时，会注册I2C1或者I2C2等，在`raspi_i2c_mst_xfer`中使用时，也需要通过获取注册时的I2C实例来判断到底是哪一个，按照目前代码中的写法，应该是`bsc_num`这个字段，表征是I2C实例，需要在注册I2C总线`rt_i2c_bus_device_register`时，将相关的信息带上，目前的代码中，并未加上I2C的实例信息，可以参考其他BSP的I2C实例实现方式，在注册时，添加上I2C的实例信息，你可以尝试解决下，如果还有问题，我们继续沟通！

mysterywolf, 2022-01-05:
[image]
有两个BSP使用到了这个成员变量 得同步升级

xiaoyu4122, 2022-01-06:
[image]
感谢回复。我只是想先验证移植的可行性。所以我把I2C驱动的编译模块给跳过了，如上图所示，进行了注释。但是，在编译过程中，有一个地方报错，如下图所示：
[image]
最后，kernel8.img文件也生成了，然后我替换了boot卡里面所有的img文件，但是系统起不来。不确定是不是编译过程中的报错的问题。希望可以解答一下。同时，我也结合别的bsp工程看看I2C实例信息缺失的问题。谢谢

chenyingchun0312, 2022-01-06:
我手头没有树莓派呢，还不是很清楚，我问问看，有没有小伙伴有，如果有的话，我可以来fix掉I2C的问题，关于你说的系统跑不起来，我还不是很清楚呢！如何这边issue回复的少，你可以到论坛中提问，那里大神还是比较多的！

mysterywolf, 2022-01-06:
下面那个mem的报错 炸弹哥看一下吧 @enkiller

mysterywolf, 2022-01-06:
应该是这个pr已经在修复这个mem的问题了 https://github.com/RT-Thread/rt-thread/pull/5472

xiaoyu4122, 2022-01-06:
搞定了，系统能跑起来，但是暂时没加载i2c，这个我再看看

mysterywolf, 2022-01-07:
@chenyingchun0312 迎春 你把iic的问题修了吧，可以让楼主用树莓派测一下，这个是 #5336 没有移除干净的部分。

chenyingchun0312, 2022-01-07:
@mysterywolf  好的，我跟涛哥借了一个树莓派！

chenyingchun0312, 2022-01-11:
准备开搞，发现已有大牛fix掉了I2C的问题，PR: https://github.com/RT-Thread/rt-thread/pull/5484

closed 2022-01-13 by mysterywolf