RT-Thread/rt-thread #5485: .\build\keil\Obj\rt-thread.axf: Error: L6200E: Symbol calloc multiply defined (by mem_std.o and syscall_mem.o).

chunyexixiaoyu, 2022-01-07:
最新版本，bsp stm32f407-atk   rtconfig.h 头文件如下：
`#ifndef RT_CONFIG_H__
#define RT_CONFIG_H__

/* Automatically generated file; DO NOT EDIT. */
/* RT-Thread Configuration */

/* RT-Thread Kernel */

#define RT_NAME_MAX 8
#define RT_ALIGN_SIZE 4
#define RT_THREAD_PRIORITY_32
#define RT_THREAD_PRIORITY_MAX 32
#define RT_TICK_PER_SECOND 1000
#define RT_USING_OVERFLOW_CHECK
#define RT_USING_HOOK
#define RT_HOOK_USING_FUNC_PTR
#define RT_USING_IDLE_HOOK
#define RT_IDLE_HOOK_LIST_SIZE 4
#define IDLE_THREAD_STACK_SIZE 1024

/* kservice optimization */

#define RT_DEBUG
#define RT_DEBUG_COLOR

/* Inter-Thread communication */

#define RT_USING_SEMAPHORE
#define RT_USING_MUTEX
#define RT_USING_EVENT
#define RT_USING_MAILBOX
#define RT_USING_MESSAGEQUEUE

/* Memory Management */

#define RT_USING_MEMPOOL
#define RT_USING_SMALL_MEM
#define RT_USING_MEMHEAP
#define RT_MEMHEAP_FAST_MODE
#define RT_USING_SMALL_MEM_AS_HEAP
#define RT_USING_HEAP

/* Kernel Device Object */

#define RT_USING_DEVICE
#define RT_USING_CONSOLE
#define RT_CONSOLEBUF_SIZE 128
#define RT_CONSOLE_DEVICE_NAME "uart1"
#define RT_VER_NUM 0x40100
#define ARCH_ARM
#define RT_USING_CPU_FFS
#define ARCH_ARM_CORTEX_M
#define ARCH_ARM_CORTEX_M4

/* RT-Thread Components */

#define RT_USING_COMPONENTS_INIT
#define RT_USING_USER_MAIN
#define RT_MAIN_THREAD_STACK_SIZE 2048
#define RT_MAIN_THREAD_PRIORITY 10

/* C++ features */

/* Command shell */

#define RT_USING_FINSH
#define RT_USING_MSH
#define FINSH_USING_MSH
#define FINSH_THREAD_NAME "tshell"
#define FINSH_THREAD_PRIORITY 20
#define FINSH_THREAD_STACK_SIZE 4096
#define FINSH_USING_HISTORY
#define FINSH_HISTORY_LINES 5
#define FINSH_USING_SYMTAB
#define FINSH_CMD_SIZE 80
#define MSH_USING_BUILT_IN_COMMANDS
#define FINSH_USING_DESCRIPTION
#define FINSH_ARG_MAX 10

/* Device virtual file system */

/* Device Drivers */

#define RT_USING_DEVICE_IPC
#define RT_USING_SERIAL
#define RT_USING_SERIAL_V1
#define RT_SERIAL_USING_DMA
#define RT_SERIAL_RB_BUFSZ 64
#define RT_USING_I2C
#define RT_USING_I2C_BITOPS
#define RT_USING_PIN

/* Using USB */

/* POSIX layer and C standard library */

#define RT_LIBC_DEFAULT_TIMEZONE 8

/* POSIX (Portable Operating System Interface) layer */

/* Interprocess Communication (IPC) */

/* Socket is in the 'Network' category */

/* Network */

/* Socket abstraction layer */

/* Network interface device */

/* light weight TCP/IP stack */

/* AT commands */

/* VBUS(Virtual Software BUS) */

/* Utilities */

/* RT-Thread online packages */

/* IoT - internet of things */

/* Wi-Fi */

/* Marvell WiFi */

/* Wiced WiFi */

/* IoT Cloud */

/* security packages */

/* language packages */

/* multimedia packages */

/* LVGL: powerful and easy-to-use embedded GUI library */

/* u8g2: a monochrome graphic library */

/* tools packages */

/* system packages */

/* acceleration: Assembly language or algorithmic acceleration packages */

/* Micrium: Micrium software products porting for RT-Thread */

/* peripheral libraries and drivers */

/* AI packages */

/* miscellaneous packages */

/* samples: kernel and components samples */

/* entertainment: terminal games and other interesting software packages */

#define SOC_FAMILY_STM32
#define SOC_SERIES_STM32F4

/* Hardware Drivers Config */

#define SOC_STM32F407ZG

/* Onboard Peripheral Drivers */

#define BSP_USING_USB_TO_USART
#define BSP_USING_SRAM
#define BSP_USING_MCU_LCD
#define BSP_USING_MCU_LCD_TEST

/* Enable File System */

/* On-chip Peripheral Drivers */

#define BSP_USING_GPIO
#define BSP_USING_UART
#define BSP_USING_UART1
#define BSP_USING_I2C2
#define BSP_I2C2_SCL_PIN 16
#define BSP_I2C2_SDA_PIN 91
#define BSP_USING_EXT_FMC_IO
#define BSP_USING_FMC

/* Board extended module Drivers */

#endif
`

编译失败：
[image]

mysterywolf, 2022-01-08:
没有复现这个问题，在rtt最新版里，没有mem_std.c这个文件。编译正常通过。

closed 2022-01-10 by mysterywolf

chunyexixiaoyu, 2022-01-10:
> 没有复现这个问题，在rtt最新版里，没有mem_std.c这个文件。编译正常通过。

这个难道是keil的？

chunyexixiaoyu, 2022-01-10:
> 没有复现这个问题，在rtt最新版里，没有mem_std.c这个文件。编译正常通过。

我是生成mdk5 发现编译无法通过

mysterywolf, 2022-01-10:
整个rtt文件夹里没有mem_std.c这个文件，你要不重新下载一份试试？这个文件早就被移除掉了，你应该是切分支或者fetch的时候没清理干净

chunyexixiaoyu, 2022-01-10:
> 整个rtt文件夹里没有mem_std.c这个文件，你要不重新下载一份试试？这个文件早就被移除掉了，你应该是切分支或者fetch的时候没清理干净

[image]

mysterywolf, 2022-01-10:
https://github.com/RT-Thread/rt-thread/tree/master/components/libc/compilers/armlibc
你看看哪有这个文件呀，这两个文件是不可能同时存在的，因为这两个文件是重命名的关系。你应该是本地拉取切分支的时候出的问题