RT-Thread/rt-thread #5486: 使用menuconfig打开lwip,webclient,mbedtls后，编译很多的error和warning。

evan-liyb, 2022-01-07:
在BSP RA6M4-CPK目录下，使用menuconfig添加了标题中使用的库后，出现很多的error和warnings。表明新版本和之前的软件包存在一定的兼容问题。如工程文件，和include路径等需要再修改一下。PS:在添加网络有关的软件包之前编译是正常的。
[image]
[image]
[image]

mysterywolf, 2022-01-07:
你用的是master吗？还是4.0.5？

mysterywolf, 2022-01-07:
不要使用master开发，master现在正在开发4.1.0版本，如果你对我们的改动不清楚的话，建议使用发布版本。webclient这些软件包报错的问题已经修正了，应该是没有打版本，你把软件包调整到latest version就好了

closed 2022-01-09 by mysterywolf

evan-liyb, 2022-01-10:
目前我要的bsp只有master才支持，所以用的master，我现在也只是处于验证阶段。提出这个issue只是为了如果此问题并非我配置导致的，在正式发布4.1.0时，注意修复一下。

mysterywolf, 2022-01-10:
你说的问题已经修复了，请选择最新版本的软件包 latest version

evan-liyb, 2022-01-10:
我看到了，目前一切正常。