RT-Thread/rt-thread #5623: the bug _timer_list_next_timeout return next timer‘s timeout_tick

thewon86, 2022-03-01:
`_timer_list_next_timeout` 函数找到定时器列表上最先达到定时时间的定时器的 `timeout_tick` 值，如果定时器列表是空返回 `RT_TICK_MAX`，也就是 0xFFFFFFFF

这样是有问题的。

`rt_timer` 的 `init_tick` 取值范围在 [0, RT_TICK_MAX] 范围内，`rt_timer_start` 可能出现在任意 `rt_tick_get()` tick 值时候，这样 `rt_timer` 的 `timeout_tick` 取值范围**也在 [0, RT_TICK_MAX] 范围内**。也就是说正常可以出现 `timeout_tick == RT_TICK_MAX` 的结果。

如此一来，无论定时器列表是不是空，都返回了 `RT_TICK_MAX` 。造成定时器线程误判为没有定时器需求，进入永久睡眠，导致所有软定时器停止工作（除非有新的定时器加入唤醒定时器线程，但是之前的定时器可能都不正常工作了）。

luguan1997, 2022-03-03:
还是注明下这个问题发现出处吧，https://club.rt-thread.org/ask/question/434783.html
这是内核部分的问题，当时我觉得有点不可思议，就先发在论坛里想讨论下。
看到别人也发了这个 issue，看来不止我一个人觉得这部分有问题。
这问题挺严重的，对于我的公司的产品来说。我司用的 soft timer 里面的代码涉及控制功率模块，造成海上的模块炸机，后续加上维护的损失还不小。
这个“问题”还是希望能赶紧确认下再修复吧。

mysterywolf, 2022-03-03:
感谢反馈！

Guozhanxin, 2022-03-07:
https://github.com/RT-Thread/rt-thread/pull/5637 修复了，你们看下。

closed 2022-03-10 by mysterywolf