RT-Thread/rt-thread #5641: cortex-m33 gcc支持

yanmowudi, 2022-03-07:
使用env工具的gcc工具链编译cortex-m33内核芯片提示arm-none-eabi-gcc: error: unrecognized argument in option '-mcpu=cortex-m33'
是否是env的gcc版本不支持cortex-m33内核？
[image]

balanceTWK, 2022-03-07:
是的。

把 env 环境变量 RTT_EXEC_PATH  改成最新 gcc 工具链路径。 

或者 改 rtconfig.py 里的内容， gcc 工具链路径改下。

balanceTWK, 2022-03-07:
另外问一句， m33 的 trustzone，咋用起来的？

yanmowudi, 2022-03-07:
> 

env工具内置的gcc会更新吗？

yanmowudi, 2022-03-07:
> 另外问一句， m33 的 trustzone，咋用的？

我是用的芯片没有TZ

yanmowudi, 2022-03-07:
> 是的。
> 
> 把 env 环境变量 RTT_EXEC_PATH 改成最新 gcc 工具链路径。
> 
> 或者 改 rtconfig.py 里的内容， gcc 工具链路径改下。

env工具里的gcc后续会更新吗？

yanmowudi, 2022-03-07:
> 另外问一句， m33 的 trustzone，咋用的？

我使用的芯片没有TZ.

wendal, 2022-03-07:
默认的gcc的版本的确太低了, 现在看来

BernardXiong, 2022-03-07:
好像后面是要发新的env版本了，发新版本可以考虑升级gcc工具链，或者可以把工具链配置成可选择/可安装的模式

closed 2022-03-08 by yanmowudi

smarthobee612, 2024-09-17:
是否解决这个问题，我也遇到编译U5A5出现：
AR19DD~1.EXE: error: unrecognized argument in option '-mcpu=cortex-m33'

suntong30, 2025-03-22:
> 是否解决这个问题，我也遇到编译U5A5出现： AR19DD~1.EXE: error: unrecognized argument in option '-mcpu=cortex-m33'

您好，请问rt-thread能支持UA5A使用tf-m吗？