RT-Thread/rt-thread #5654: [STM32]drv_eth.c在H7系列报错

CharlesXSong, 2022-03-10:
### 初始化函数rt_stm32_eth_init
rt_stm32_eth_init初始化EthHandle时的代码如下：
```c
    __HAL_RCC_ETH_CLK_ENABLE();

    phy_reset();

    /* ETHERNET Configuration */
    EthHandle.Instance = ETH;
    EthHandle.Init.MACAddr = (rt_uint8_t *)&stm32_eth_device.dev_addr[0];
    EthHandle.Init.AutoNegotiation = ETH_AUTONEGOTIATION_DISABLE;
    EthHandle.Init.Speed = ETH_SPEED_100M;
    EthHandle.Init.DuplexMode = ETH_MODE_FULLDUPLEX;
    EthHandle.Init.MediaInterface = ETH_MEDIA_INTERFACE_RMII;
    EthHandle.Init.RxMode = ETH_RXINTERRUPT_MODE;
#ifdef RT_LWIP_USING_HW_CHECKSUM
    EthHandle.Init.ChecksumMode = ETH_CHECKSUM_BY_HARDWARE;
#else
    EthHandle.Init.ChecksumMode = ETH_CHECKSUM_BY_SOFTWARE;
#endif

    HAL_ETH_DeInit(&EthHandle);
```

然而，在`bsp\stm32\libraries\STM32H7xx_HAL\STM32H7xx_HAL_Driver\Inc\stm32h7xx_hal_eth.h`中关于该句柄Init成员`  ETH_InitTypeDef            Init;                      /*!< Ethernet Init Configuration */`的定义如下：
```c
typedef struct
{
   
  uint8_t                     *MACAddr;                  /*!< MAC Address of used Hardware: must be pointer on an array of 6 bytes */
  	
  ETH_MediaInterfaceTypeDef   MediaInterface;            /*!< Selects the MII interface or the RMII interface. */

  ETH_DMADescTypeDef          *TxDesc;                   /*!< Provides the address of the first DMA Tx descriptor in the list */
  
  ETH_DMADescTypeDef          *RxDesc;                   /*!< Provides the address of the first DMA Rx descriptor in the list */
  
  uint32_t                    RxBuffLen;                 /*!< Provides the length of Rx buffers size */

}ETH_InitTypeDef;
```
可以看到驱动代码初始化的属性和H7对应的结构体成员不匹配，导致在H7中使用LWIP和ETH功能时报错

### 收发线程

`rt_stm32_eth_tx`和`rt_stm32_eth_rx`也会因为没有`EthHandle.TxDesc`等成员与之对应而报错：
`bsp\stm32\libraries\STM32H7xx_HAL\STM32H7xx_HAL_Driver\Inc\stm32h7xx_hal_eth.h`
```c
#if (USE_HAL_ETH_REGISTER_CALLBACKS == 1)
typedef struct __ETH_HandleTypeDef
#else
typedef struct
#endif
{
  ETH_TypeDef                *Instance;                 /*!< Register base address       */
  
  ETH_InitTypeDef            Init;                      /*!< Ethernet Init Configuration */
	
  ETH_TxDescListTypeDef      TxDescList;                /*!< Tx descriptor wrapper: holds all Tx descriptors list 
                                                            addresses and current descriptor index  */
		
  ETH_RxDescListTypeDef      RxDescList;                /*!< Rx descriptor wrapper: holds all Rx descriptors list 
                                                            addresses and current descriptor index  */ 
  
  HAL_LockTypeDef            Lock;                      /*!< Locking object             */
 
  __IO HAL_ETH_StateTypeDef  gState;                   /*!< ETH state information related to global Handle management 
                                                              and also related to Tx operations.
                                                             This parameter can be a value of @ref HAL_ETH_StateTypeDef */
  
  __IO HAL_ETH_StateTypeDef  RxState;                   /*!< ETH state information related to Rx operations.
                                                             This parameter can be a value of @ref HAL_ETH_StateTypeDef */
  
  __IO uint32_t              ErrorCode;                 /*!< Holds the global Error code of the ETH HAL status machine 
                                                             This parameter can be a value of of @ref ETH_Error_Code */
  
  __IO uint32_t              DMAErrorCode;              /*!< Holds the DMA Rx Tx Error code when a DMA AIS interrupt occurs
                                                             This parameter can be a combination of @ref ETH_DMA_Status_Flags */
  
  __IO uint32_t              MACErrorCode;              /*!< Holds the MAC Rx Tx Error code when a MAC Rx or Tx status interrupt occurs
                                                             This parameter can be a combination of @ref ETH_MAC_Rx_Tx_Status */ 
    
  __IO uint32_t              MACWakeUpEvent;            /*!< Holds the Wake Up event when the MAC exit the power down mode
                                                             This parameter can be a value of @ref ETH_MAC_Wake_Up_Event */
    
  __IO uint32_t              MACLPIEvent;               /*!< Holds the LPI event when the an LPI status interrupt occurs.
                                                             This parameter can be a value of @ref ETHEx_LPI_Event */
                                                             
#if (USE_HAL_ETH_REGISTER_CALLBACKS == 1)

  void    (* TxCpltCallback)     ( struct __ETH_HandleTypeDef * heth);   /*!< ETH Tx Complete Callback */
  void    (* RxCpltCallback)     ( struct __ETH_HandleTypeDef * heth);  /*!< ETH Rx  Complete Callback     */
  void    (* DMAErrorCallback)   ( struct __ETH_HandleTypeDef * heth);  /*!< ETH DMA Error Callback   */
  void    (* MACErrorCallback)   ( struct __ETH_HandleTypeDef * heth);  /*!< ETH MAC Error Callback     */
  void    (* PMTCallback)        ( struct __ETH_HandleTypeDef * heth);  /*!< ETH Power Management Callback            */
  void    (* EEECallback)        ( struct __ETH_HandleTypeDef * heth);  /*!< ETH EEE Callback   */
  void    (* WakeUpCallback)     ( struct __ETH_HandleTypeDef * heth);  /*!< ETH Wake UP Callback   */

  void    (* MspInitCallback)    ( struct __ETH_HandleTypeDef * heth);    /*!< ETH Msp Init callback              */
  void    (* MspDeInitCallback)  ( struct __ETH_HandleTypeDef * heth);    /*!< ETH Msp DeInit callback            */

#endif  /* USE_HAL_ETH_REGISTER_CALLBACKS */                                                             

} ETH_HandleTypeDef;
```

xiangxistu, 2022-03-10:
可以参考一下 stm32H750-artpi 的处理方式

LXGMAX, 2023-09-13:
直接用ART-Pi的代码
https://github.com/RT-Thread-Studio/sdk-bsp-stm32h750-realthread-artpi/blob/master/libraries/drivers/drv_eth.c

closed 2025-09-25 by Rbb666