RT-Thread/rt-thread #5668: [STM32F2](drv_eth.c) LAN8720A 可以发送 数据，无法接收到

jiuxiaxixi, 2022-03-12:
## 1.环境 使用RT-Studio 创建的标准F2 工程，已经修改了HSE时钟。
 编译不通过，提示 重复定义的 `PHY_FULL_DUPLEX`
 
修改 `PHY_FULL_DUPLEX`的名称为 `PHY_FULL_DUPLEX`
编译通过，但是不知道为啥

## 2. phy_reset.c 如下
```c
#include "board.h"
#include "drv_common.h"

#include <rtdevice.h>

#define RESET_IO GET_PIN(A,3)

void phy_reset(void){
    rt_pin_write(RESET_IO, PIN_LOW);
    rt_thread_mdelay(50);
    rt_pin_write(RESET_IO, PIN_HIGH);
}

int phy_init(void){
    rt_pin_mode(RESET_IO, PIN_MODE_OUTPUT);
    rt_pin_write(RESET_IO, PIN_HIGH);
    return RT_EOK;
}

INIT_APP_EXPORT(phy_init);
```
## 开打 #define DRV_DEBUG 日志如下

```
[I/sal.skt] Socket Abstraction Layer initialize success.
msh >[D/drv.emac] Found a phy, address:0x00
[D/drv.emac] RESET PHY!
[D/drv.emac] phy basic status reg is 0x7829
[D/drv.emac] phy control status reg is 0x1058
[D/drv.emac] link up
[D/drv.emac] 100Mbps
[D/drv.emac] full-duplex
[D/drv.emac] transmit frame length :42
[D/drv.emac] receive frame faild
[D/drv.emac] phy basic status reg is 0x782D
[D/drv.emac] phy control status reg is 0x1058
[D/drv.emac] phy basic status reg is 0x782D
```

定位到 
```c
  /* Get received frame */
    state = HAL_ETH_GetReceivedFrame_IT(&EthHandle);
    if (state != HAL_OK)
    {
        LOG_D("receive frame faild");
        return NULL;
    }
```
HAL_ETH_GetReceivedFrame_IT(&EthHandle); 返回了 HAL_ERROR
```c
  /* Scan descriptors owned by CPU */
  while (((heth->RxDesc->Status & ETH_DMARXDESC_OWN) == (uint32_t)RESET) && (descriptorscancounter < ETH_RXBUFNB))
```
这个条件没有过，

请问有解决的方法么？

芯片是STM32F207ZET6,原理图如下

[image]

jiuxiaxixi, 2022-03-12:
@whj4674672  可以帮忙看看么？

jiuxiaxixi, 2022-03-12:
## STM32 F4的寄存器状态
[image]
## STM32 F2的寄存器状态
[image]

jiuxiaxixi, 2022-03-12:
[image]

STM32F2的 OWM bit 是SET 的状态。其他位好像都是无效的
OWN: Own bit
When set, this bit indicates that the descriptor is owned by the DMA. When this bit is reset, it
indicates that the descriptor is owned by the CPU. The DMA clears this bit either when it
completes the frame transmission or when the buffers allocated in the descriptor are read
completely. The ownership bit of the frame’s first descriptor must be set after all subsequent
descriptors belonging to the same frame have been set.

jiuxiaxixi, 2022-03-12:
好像是DMA 没有释放掉的意思，但是不清楚是软件的问题还是硬件的问题

jiuxiaxixi, 2022-03-12:
这个是用STM32 ( 192.168.1.10) ping 电脑主机的IP( 192.168.1.23)
[image]
电脑回复了，但是32 一直无法收到

aozima, 2022-03-12:
有没可能  硬件RX 线没焊好？ MAC地址没设置对？...

mysterywolf, 2022-03-12:
建议到论坛提问 会有更多人来回复

jiuxiaxixi, 2022-03-12:
> 

Mac地址怎么设置呀！ 我看到它发出来是有mac地址的

jiuxiaxixi, 2022-03-12:
> 建议到论坛提问 会有更多人来回复

好的 下次一定

mysterywolf, 2022-03-14:
请到论坛提问 issue是用来反馈bug的，在这里提问别人看不见的。

closed 2022-03-14 by mysterywolf

gamingjiang, 2023-09-08:
老哥 解决了吗  什么原因啊   我和你一样的问题

jiuxiaxixi, 2023-09-12:
没有解决啊， 忘记了