RT-Thread/rt-thread #5688: NK980IOT v1.0疑似硬件问题，请协助

shannon2000, 2022-03-18:
首先确认，板子的CDC驱动(CDC驱动WIN10 64提升已经内置，所以跳过安装)、WinUSB4NuVCOM都安装了，检测方式：通过USB线接virtualCOM口，上电后可以确认串口有log输出，但是日志有时候是“NUC980 IBR 20180813”，有时候是“NUC980 IBR 2018081”，具体驱动安装和串口打印如下图。
[https://oss-club.rt-thread.org/uploads/20220317/ffd129e60a748573192255563d12996f.png](url)
[https://oss-club.rt-thread.org/uploads/20220317/b5e9c62b4a954fb842d11056c337e715.png](url)
其二确认USB线没有问题，通过第一步的接virtualCOM的串口打印可以知道，USB线功能正常。
其三排除电源影响，外接DC 5V电源。
其四排除boot mode弄错了，拨码开关的四种模式都测试了。
[https://oss-club.rt-thread.org/uploads/20220317/f2da2af9c631d105a5b8311d53643232.jpg.webp](url)
目前的表现就是：在外接5V DC电源的情况下，接virtualCOM，然后打开NuWriter，选择“980DK61Y”,不能自动发现板子，强行CONNECT就跳出如下对话框。
[https://oss-club.rt-thread.org/uploads/20220317/54b101dfd4a379fecdb32d4bc82a9f47.png](url)
目前手头有万用电表，能否告诉检测下哪个部位确认HOST USB0是否坏了？

后续按照热心网友建议，选择USB启动模式，只插CON4，不插CON9和DC-IN，检查串口输出。
按了几次reset，串口打印如下，字符串只有NUC980，没有boot from XXX的内容。
`NUC980 IBR 201808
NUC980 IBR 201
NUC980 IBR 2018081
NUC980 IBR 2018081`
后续电表测量USB0_ID的电压3.51V。

shannon2000, 2022-03-18:
问题再RT-THREAD论坛的地址：https://club.rt-thread.org/ask/question/435015.html

BernardXiong, 2022-03-24:
怎么硬件相关的问题也会发过来了

closed 2022-03-24 by BernardXiong