RT-Thread/rt-thread #5708: [BUG]simulator下创建文件失败&使用windows的socket失败

BreederBai, 2022-03-24:
我的[多旋翼飞控程序](https://gitee.com/breederbai/grapilot)使用rt-thread，最近我将rtthread升级到4.1.0版本，发现无法创建文件并且无法[创建socket](https://gitee.com/breederbai/grapilot/blob/os-update/platform/simulator/windows/drivers/gp_socket_udp.c#L88)。
### 该问题是怎么引起的？
使用simulator bsp，创建文件失败，创建socket也失败
### 重现步骤
`#define PARAM_FILE_NAME "/sys/param.xml"
fd = open(PARAM_FILE_NAME, O_WRONLY | O_CREAT);
`
fd的值为-1。
[代码链接](https://gitee.com/breederbai/grapilot/blob/os-update/libraries/parameter/param.c#L1581)

mysterywolf, 2022-03-24:
simulator BSP现在不支持仿真网络，如果想要仿真网络需要用QEMU模拟器，
这里有视频讲解怎么用qemu模拟器的文件系统和网络功能
https://www.bilibili.com/video/BV1TJ411r7dX?p=1

mysterywolf, 2022-03-24:
simulator的BSP应该是支持文件系统的，但是我不知道怎么用，我会协调一下，出个文档来。

BreederBai, 2022-03-25:
@mysterywolf 好的，谢谢您的回复。网络和文件系统，我用之前的4.0.0版本是没有问题的。socket直接调用windows的库，文件系统的使用和在STM32中的一样。上述的代码，使用旧版本是没有问题的

mysterywolf, 2022-03-25:
可以帮忙协助查找一下问题嘛

BreederBai, 2022-03-25:
@mysterywolf  可以，具体需要怎么做

BreederBai, 2022-03-25:
我跟踪代码到地址转换都是正确的，在调用_open打开文件时，返回了ERROR_PATH_NOT_FOUND。这个就很奇怪。
传入“/sys/param.xml”，地址转换完为“.\\disk\\sys\\param.xml”
[image]
我看有人说和什么字符宽度有关系，我不太明白这个字符宽度是啥意思，是不是指字符的编码不一样

BreederBai, 2022-03-29:
@mysterywolf 还需要我做什么吗？

mysterywolf, 2022-03-29:
OK

BreederBai, 2022-03-30:
@mysterywolf 我似乎找到了问题，fcntl.h文件中的宏定义看着像是16进制的，但前面没有0x，我改为下图这样，似乎可以创建读取文件了。
[image]

mysterywolf, 2022-03-30:
感谢 我4.1.1 统一一下编码，现在存在两套编码，需要统一成一套就好了。这个纯数字的是musl库的标准 十六进制是newlib的标准

mysterywolf, 2022-03-30:
这两天我修一下这个编码的问题 统一改成十六进制，八进制的编码就是个奇葩。然后你再试试

mysterywolf, 2022-03-30:
https://github.com/RT-Thread/rt-thread/pull/5267 这个pr已经修了一部分了 但是没有考虑到simulator在windows下的行为

BreederBai, 2022-03-30:
@mysterywolf 还有个问题，O_BINARY这个宏定义似乎没有起作用或`fstat(fd,&sta_f);`函数有些问题。我使用fstat来读取文件大小，返回的文件大小的回车是\r\n，但实际用read函数读出的文件大小的\n为回车的。我用vscode看文件的时候，显示是\r\n结尾的，这个问题感觉有点奇怪。
代码链接：https://gitee.com/breederbai/grapilot/blob/os-update/libraries/parameter/param.c#L1678
代码写入出问题的文件（右下角有回车的编码）：
[新建 文本文档 (2).txt](https://github.com/RT-Thread/rt-thread/files/8381939/2.txt)
[image]
之前代码创建的文件（右下角有回车的编码）：
[image]

mysterywolf, 2022-03-30:
O_BINARY 估计也是和编码有关系，你编码改的有问题，所以导致O_BINARY不会被识别到，等我改好我通知你再测一下。

BreederBai, 2022-03-31:
@mysterywolf 好的，期待！

BreederBai, 2022-03-31:
@mysterywolf 问一个不相关的问题，rtthread能不能在ubuntu下运行，不使用qemu的情况下

mysterywolf, 2022-03-31:
linux的问题你再开一个issue吧 这个issue我来记录一下simulator的问题。下个礼拜我会去着手修复，这个礼拜很忙。

BreederBai, 2022-04-01:
@mysterywolf 好的

mysterywolf, 2022-04-05:
你好 用邮件ping我一下我的邮箱 我加你一下微信

BreederBai, 2022-04-06:
好的

BreederBai, 2022-04-07:
@mysterywolf 我试了下似乎还是不行
[image]

BreederBai, 2022-04-09:
@mysterywolf 您那边是如何测试这个文件系统的？能否给个例子，我看看是不是我的代码有问题。ps：我用的visual studio 2019

mysterywolf, 2022-04-09:
https://github.com/RT-Thread/rt-thread/pull/5793

mysterywolf, 2022-04-09:
刚刚修复好 等合并之后测试一下

BreederBai, 2022-04-09:
好的

mysterywolf, 2022-04-11:
感谢反馈！

BreederBai, 2022-04-13:
@mysterywolf 我试了下，还是有问题，创建依然会失败，下图是最新版本的。
[image]

BreederBai, 2022-04-13:
@mysterywolf 是不是O_CREAT宏定义的值有问题，在windows里面是其他的值。

BreederBai, 2022-04-13:
@mysterywolf 更改路径后，可以写入文件，但是小写全部变成了大写
[image]

mysterywolf, 2022-04-13:
请把测试代码传上来一下吧

BreederBai, 2022-04-13:
打开文件时会有一个没有RTC的警告
[image]

BreederBai, 2022-04-13:
@mysterywolf 我还有一个疑问，root和sd都从初始化成功了。程序时如何选择写入到那个文件系统的

mysterywolf, 2022-04-13:
找不到RTC倒是没什么问题 没开RTC 写入文件的时间就是1970年而已
FATFS挂载在sd目录下，你往sd文件夹路径下写就是往fatfs写，假如说，有另一个文件系统，比如littlefs，挂在了flash文件夹下，那你往flash文件夹路径下写，就是访问littlefs。这个和类unix系统是一致的，如果你平时用windows系统的话可能比较难理解。

mysterywolf, 2022-04-13:
可以看一下rt-thread的文档中心

BreederBai, 2022-04-13:
root是挂载在/下的，如果我直接往根目录先创建文件，理论上应该也是有的。而且我之前的使用，他确实是有的。之前不能用的时候，我把打开文件的宏定义之前在前面增加0x变成16进制，也是能写到rt-thread\bsp\simulator\disk这个目录下的。我感觉这个地方还是有些问题。有没有可能是不同平台宏定义的值不一样。类unix和windows中不一样。

BreederBai, 2022-04-13:
在fatfs下创建文件、写入这些，是没问题的。在飞控硬件上跑，是可以存到sd卡上的。

mysterywolf, 2022-04-22:
测试例程：
```c
#include <rtthread.h>
#include <unistd.h>
#include <fcntl.h>

#define PARAM_FILE_NAME                "/sd/sys/param.xml"

int main(void)
{
    rt_thread_mdelay(1000);
    int fd;
    fd = open(PARAM_FILE_NAME, O_WRONLY | O_CREAT | O_APPEND);

    if (fd < 0) {
        rt_kprintf("creat file fail, fd %d!\n", fd);
    }
    else {
        char buffer[256];
        rt_memset(buffer, 0x31, sizeof(buffer));
        write(fd, buffer, sizeof(buffer));
        fsync(fd);
        close(fd);
    }

    int o_create = O_CREAT;
    rt_kprintf("O_CREAT is %X!\n", o_create);

    rt_kprintf("Hello RT-Thread!\n");
    return 0;
 }

```

closed 2022-04-24 by mysterywolf