RT-Thread/rt-thread #5715: A locking problem

ryancaicse, 2022-03-25:
Hi, it seems the lock is not released before destroying, which is undefined behavior.
https://github.com/RT-Thread/rt-thread/blob/9bbcd1a8b0b5b7a4892690680234054b4d6e0b58/bsp/simulator/drivers/sdl_fb.c#L520-L523

mysterywolf, 2022-03-25:
Thank you for reporting that,

ryancaicse, 2022-03-25:
@mysterywolf Hi, thank you for your reply.

While fixing this, I suggest that it could be better to use a loop to enforce the happens-before order between the waiting and signaling. Otherwise, it is possible for the thread to signal first and, later, the thread goes to the waiting site forever without signal notified anymore.

https://github.com/RT-Thread/rt-thread/blob/9bbcd1a8b0b5b7a4892690680234054b4d6e0b58/bsp/simulator/drivers/sdl_fb.c#L274-L286

https://github.com/RT-Thread/rt-thread/blob/9bbcd1a8b0b5b7a4892690680234054b4d6e0b58/bsp/simulator/drivers/sdl_fb.c#L520-L522

**the recommended usage of thread_cond_signal and thread_cond_wait would be**

```
pthread_mutex_t count_lock;
pthread_cond_t count_nonzero;
unsigned count;

decrement_count()
{
    pthread_mutex_lock(&count_lock);
    while (count == 0)
        pthread_cond_wait(&count_nonzero, &count_lock);
    count = count - 1;
    pthread_mutex_unlock(&count_lock);
}

increment_count()
{
    pthread_mutex_lock(&count_lock);
    if (count == 0)
        pthread_cond_signal(&count_nonzero);
    count = count + 1;
    pthread_mutex_unlock(&count_lock);
}
```

ryancaicse, 2022-03-25:
Just like this
https://github.com/RT-Thread/rt-thread/blob/9bc3896e41716f6415e948ffea153a5453fe4b34/examples/libc/ex2.c#L43-L55

mysterywolf, 2022-03-25:
Thanks for your suggestion we may fix this problem when release 4.1.0

xiangxistu, 2022-08-03:
https://github.com/RT-Thread/rt-thread/pull/6113

Wakeup loss is a secret error, this PR can fix this problem.
happens-before is more effective fix way, Thank you for your suggestion.
----
But in this issue, we just need unlock this "sdl_ok_mutex".

closed 2022-08-03 by mysterywolf