RT-Thread/rt-thread #5751: HPM6750  rtt4.05 移植到 ses stdio下， 进入rtt(bm状态)之前 printf可以打印浮点数， 进入到rtt线程后 再次调用printf 打印浮点数会死机。

yandld, 2022-03-31:
问题描述：HPM6750  rtt4.05 移植到 ses stdio下， 进入rtt(bm状态)之前 printf可以打印浮点数， 进入到rtt线程后 再次调用printf 打印浮点数会死机。
<由于不能贴图，详见这里： >
https://note.youdao.com/s/YBeuWpLY

复现方法： 
1. 使用rtt4.05 放到ses下，把libcpu中的文件对应做移植，rtt成功跑起来， ses O0, 不开启dfpu(详见附件链接)
2. 在进入rtt系统之前可以正常打印浮点：

3.进入thrad之后打印浮点会死机，打印整数正常：

其中rt_kprintf已经被我改成vsnprintf,所以支持浮点：

我试了其他内置库函数，比如sprintf, 只要是浮点数，在thread里一律死机，不使用浮点，一切正常。

猜测原因：  rtt4.05移植到ses stdio下对于浮点数现成压栈出栈的移植问题。
请教各位大神可否帮忙解决。。

整个工程打包及链接：https://note.youdao.com/s/YBeuWpLY

北京超核电子科技
20220331

mysterywolf, 2022-03-31:
你好 不要使用printf vsnprintf snprintf打印浮点数，请使用rt_kprintf rt_vsnprintf rt_snprintf代替，并安装该软件包：https://github.com/mysterywolf/rt_vsnprintf_full
4.0.5 下需要手动处理一下，将kservice.c中原来的rt_vsnprintf注释掉，用这个软件包的函数代替。4.1.0可以不需要手动处理。

yandld, 2022-04-01:
您好，感谢回答，我尝试了这个方案，但是并没有改善，应该不是这个问题。
最后我把rt_kprintf里面的东西全部注释掉，空跑一个rt_kprintf, 竟然还是会死掉，感觉还是和浮点数的thread压栈出栈有关，或者和ses的编译器的问题有关感觉。

mysterywolf, 2022-04-01:
嗯 那应该是和出入栈相关的了

yandld, 2022-04-01:
是的，浮点数的线程输入栈有关系。
我再我的浮点运算算法调用前后加 rt_hw_interrupt_disable();/enable() 就正常了。
SES6.22 O0, 不开dfpu

mysterywolf, 2022-04-01:
如果是这样子的话，应该是任何一个带有浮点形参的函数都会出问题的。尤其是double类型，是否是FPU的问题。
建议到论坛提问。

BernardXiong, 2022-04-03:
对于新版本的SES不太了解，老版本上SES有问题的：

SES会用自有的一份工具链，那份工具中的libc应该是一个叫crossworks的libc，和newlib有挺大的一个区别。所以把RTT直接用于SES中会存在问题，**不能够得到支持**。

不知道新版本是否有改善。

mysterywolf, 2022-04-08:
非系统bug问题，请移步论坛

closed 2022-04-08 by mysterywolf

yandld, 2022-04-13:
谢谢各位熊大解答，
我现在已经暂时放弃ses了，切到rtt stdio，但是问题依旧。已经联系到HPM原厂工程师正在帮忙解决，非常感谢大佬的支持。