RT-Thread/rt-thread #5799: 【BUG】 STM32F4 USBD 无法枚举

jiuxiaxixi, 2022-04-10:
开发环境
野火骄阳 STM32F407
使用Rt-thread Studio 创建STM32F407 工程, 使用4.0.2 4.0.3 4.1.0 3.15 lts
使用STM32CubeMx 创建项目 配置USB RCC
RT-thread Setting 设置
以上步骤都相同的时候，使用rt-thread lastest 就可以正常枚举USB，有熟悉的人知道 具体是那些地方发生一改变么？

已经尝试的步骤

拷贝 lastest 中的 STM32 LIB
更换 rt-thread/components/drivers/usb 中的 USB core lib
更换 driver 中的drv_usbd.c
到 4.0.2版本中都无法正常枚举？
现在非常疑惑，最新发布的4.1.0 也无法正常枚举
我已经把自己的代码移植到 lastest上了。USB可以正常枚举，所以应该不是我自己的代码引起USB出现无法枚举的问题。

jiuxiaxixi, 2022-04-10:
我反向操作了以下，把 `4.0.2` 版本的USB库拷贝到 latest 版本中，是可以正常枚举的

jiuxiaxixi, 2022-04-10:
@armink  大佬有想法么？

liukangcc, 2022-04-12:
https://club.rt-thread.org/ask/question/430005.html 

[image]

@mysterywolf 我怀疑是 Cubemx 操作有问题，你看上面那个帖子，usb cdc 就是好使得，我们 udb 的 usb 也一直是好使的

jiuxiaxixi, 2022-04-14:
> https://club.rt-thread.org/ask/question/430005.html
> 
> [image]
> 
> @mysterywolf 我怀疑是 Cubemx 操作有问题，你看上面那个帖子，usb cdc 就是好使得，我们 udb 的 usb 也一直是好使的

我吧生产的库也替换了的，还是不能枚举