RT-Thread/rt-thread #5806: [NUMAKER-IOT-M487]: rt_hwcrypto_rng_update_ctx can't generate good random numbers

foldl, 2022-04-13:
## Version

* RT-Thread Version: 4.0.4
* NUMAKER-IOT-M487 SDK: [almost newest on main branch](https://github.com/RT-Thread/rt-thread/tree/ec539bc930af1b91c6bf9758160b0150f9316c6d)

## Issue

`rt_hwcrypto_rng_update_ctx` can't generate good random numbers.
Below test fails which did a naïve check on the randomness, indicating that:

1. Generated random bytes do not follow a uniform distribution;
1. There are far more 0s than 1s on the most signification bit of each byte.

```c
void test_rng(struct rt_hwcrypto_device *device)
{
    struct rt_hwcrypto_ctx *ctx = rt_hwcrypto_rng_create(device);
    int i;
    static uint32_t counts[256] = {0};

#define CNT_THRES   700

    print("generating 256 * 1000 random bytes:\n");
    for (i = 0; i < 256 * 1000 / 4; i++)
    {
        uint32_t v = rt_hwcrypto_rng_update_ctx(ctx);
        counts[v & 0xff]++; v >>= 8;
        counts[v & 0xff]++; v >>= 8;
        counts[v & 0xff]++; v >>= 8;
        counts[v & 0xff]++;
    }

    rt_hwcrypto_rng_destroy(ctx);

    print("a simple check: if random bytes follow an uniform distribution...");
    int flag = 0;
    for (i = 0; i < 256; i++)
    {
        print("%d: %d\n", i, counts[i]);
        if (counts[i] < CNT_THRES)
            flag = 1;
    }
    print("%s\n", flag == 0 ? "PASS" : "FAIL");
}
```

Output (sample):

```
1: 1292
...
127: 1265
128: 739
...
```

@wosayttn.

wosayttn, 2022-04-14:
Hi,

Thanks the issue report. We can reproduce the issue on my side.

Root cause: M487 is without TRNG peripheral, but it enabled in default configuration. 
The TRNG driver porting checks TRNG function is available or not as below. if not, it will route to SW rand function.

```bash
rt_uint32_t nu_trng_rand(struct hwcrypto_rng *ctx)
{
    if (!s_i32TRNGEnable)
    {
        /* Use software rand */
        return (rt_uint32_t)rand();
    }

    return nu_trng_run();
}
```

Solution: To comment BSP_USING_TRNG definition in rtconfig.h temporarily and rebuild. After commenting, Crypt-PRNG module function will be enabled and act the default random generator.

We will make a PR for this issue.

foldl, 2022-04-14:
I think it is really a  bad ideal to fallback to PRNG in `nr_trng_rand`, and `nu_trng_run`:

https://github.com/RT-Thread/rt-thread/blob/efdd28bde620b7695ef852b8ab647cbddb12e646/bsp/nuvoton/libraries/m480/rtt_port/drv_trng.c#L37

foldl, 2022-04-14:
What does "Random number generator" mean ?

https://www.nuvoton.com/products/microcontrollers/arm-cortex-m4-mcus/m487-ethernet-series/

[image]

wosayttn, 2022-04-14:
Hi,
M487 supports pseudo random number generator in cryptographic accelerator (CRYPTO) peripheral only.
Pseudo random number generator is called PRNG, is different with true random number generator(TRNG) peripheral.
More details, please refer M480 TRM. Thanks.

foldl, 2022-04-14:
Hi,

It's better to update the above URL to clearly state that it is a pseudo random number generator, as there are other
M48X series that do have TRNG.

wosayttn, 2022-04-14:
Hi,
I really appreciate your valuable opinion AND also pass it to marketing team.

foldl, 2022-04-14:
PR #5812 can fix the problem, and `hwcrpyto_rng` is then backed by PRNG.

closed 2022-04-14 by foldl