RT-Thread/rt-thread #5808: RT-Thread 4.1.1 路线与意见征集

mysterywolf, 2022-04-13:
4.1.0路线和意见征集issue参见：https://github.com/RT-Thread/rt-thread/issues/5342

4.1.0已经发布，接下来将发布4.1.1版本。按照版本发布原则，4.1.1是4.1.0的bug修正版本，其修改幅度不能过于剧烈。4.1.1的计划内容包含
- 社区在使用4.1.0发布版本时反馈的问题
- 4.1.0征集意见时提出，但是未计划在4.1.0中修改的部分

jiuxiaxixi, 2022-04-14:
请修复通过RTT  STUDIO  创建的项目无法枚举USB的BUG

BernardXiong, 2022-04-15:
> 4.1.0已经发布，接下来将发布4.1.1版本。按照版本发布原则，4.1.1是4.1.0的bug修正版本，其修改幅度不能过于剧烈。4.1.1的计划内容包含：
> 
> * 社区在使用4.1.0发布版本时反馈的问题
> * 4.1.0征集意见时提出，但是未计划在4.1.0中修改的部分
> 
> 之前4.1.0路线和意见征集issue参见指路：#5342 5.0.0意见征集指路：#5616

4.1.0发布后，新特性应该都会冻结了的，主要是修bug

mysterywolf, 2022-04-19:
https://github.com/RT-Thread/rt-thread/pull/5742

mysterywolf, 2022-04-21:
rt_kprintf需要精炼和整理一下，函数里有一些没有用的参数或者代码。比如RT_PRINTF_PRECISION这个宏，必须定义，不定义会直接报错。

mysterywolf, 2022-04-26:
- [ ] https://club.rt-thread.org/ask/article/40bdad8018337ca2.html

mysterywolf, 2022-04-30:
设备框架胡乱返回RT_EOK错误码的问题：
https://github.com/RT-Thread/rt-thread/pull/5898
http://bbs.eeworld.com.cn/thread-1196113-1-1.html

mysterywolf, 2022-04-30:
[image]
设备框架没有提供init open close的接口，导致用户如果想要做一些构造和析构操作，例如init的时候初始化IO，close的时候将IO恢复到默认状态，关闭对应的时钟低功耗，都不能实现。

mysterywolf, 2022-05-05:
STM32没有输入捕获驱动：https://club.rt-thread.org/ask/question/60abfd50005a061b.html

Guozhanxin, 2022-05-13:
完善 vs 上 libc，能编译过 lwip 和sal

mysterywolf, 2022-05-17:
https://github.com/RT-Thread/rt-thread/blob/master/components/libc/posix/ipc/mqueue.c#L209

mysterywolf, 2022-05-20:
BSP:
STM32 pwm驱动在L4下 TIM17可能没有这个定时器：https://club.rt-thread.org/ask/question/0876e296d846a5e1.html

dongly, 2022-05-25:
free 可否增加实际空余内存显示 free? 每次查看都是要计算一下 free = total - used

mysterywolf, 2022-05-29:
string.h中部分glibc拓展函数各个编译平台不一致
例如strnlen 在IAR中定义 但是keil gcc下没有定义

mysterywolf, 2022-05-31:
libc 补充timelocal 和 localtime函数

mysterywolf, 2022-05-31:
## kernel 内核相关
警告消除：https://github.com/RT-Thread/rt-thread/pull/5744

断言和参数检查
https://github.com/RT-Thread/rt-thread/issues/5939
https://github.com/RT-Thread/rt-thread/pull/5850
https://github.com/RT-Thread/rt-thread/pull/5576

mysterywolf, 2022-06-06:
龙伟：Cortex M 的开关中断保护 用 basepri 这个寄存器
https://www.freertos.org/FreeRTOS_Support_Forum_Archive/February_2014/freertos_Cortex_M3_4_BASEPRI_2cd231d0j.html

mysterywolf, 2022-06-12:
建议实现 clock_nanosleep clock_gettime 像 [Xenomai](https://xenomai.org/documentation/xenomai-3/html/xeno3prm/group__cobalt__api__time.html) 这种 （https://github.com/RT-Thread/rt-thread/issues/541#issuecomment-1126845062）

mysterywolf, 2022-06-15:
https://github.com/RT-Thread/rt-thread/pull/5848#issuecomment-1117597424

thewon86, 2022-06-16:
#5876

mysterywolf, 2022-06-22:
strncpy strlcpy的问题https://blog.csdn.net/tigerjibo/article/details/6412759

mysterywolf, 2022-07-04:
https://github.com/RT-Thread/rt-thread/pull/6130

mysterywolf, 2022-07-04:
https://club.rt-thread.org/ask/question/1dfc5b29df0304a7.html

mysterywolf, 2022-07-05:
只让qemu-vepress Werror 就可以了，没有必要让其他bsp也把警告当成错误 https://github.com/RT-Thread/rt-thread/pull/5785/files

mysterywolf, 2022-07-09:
## 需要继续跟踪的PR和论坛帖子
- [ ] #5296 其他系列的STM32是否也有这个问题
- [ ] STM32 LPUART的问题
  - https://club.rt-thread.org/ask/question/433249.html RTT 针对stm32lx系列低功耗串口驱动bug
  - https://club.rt-thread.org/ask/question/424812.html 反馈rtt串口驱动对低功耗串口lpuart1不兼容的问题
  - https://club.rt-thread.org/ask/article/3412.html STM32L4 LPUART1 DMA下不能工作的解决办法
- [x] https://club.rt-thread.org/ask/question/434691.html 论坛反馈的finsh_getchar的一个bug
- [ ] https://club.rt-thread.org/ask/article/54.html STM32 UART
- [ ] https://club.rt-thread.org/ask/question/425502.html STM32L0 drv_usart.c的BUG及解决
- [x] https://club.rt-thread.org/ask/question/429786.html  ulog的问题 https://github.com/RT-Thread/rt-thread/pull/5114
- [ ] https://club.rt-thread.org/ask/question/424912.html drv_pwm.c 有bug
- [x] https://club.rt-thread.org/ask/question/434822.html 同时ulog的timestamp与rtc打印无输出 https://github.com/RT-Thread/rt-thread/pull/5632
- [ ] https://club.rt-thread.org/ask/article/3483.html n32g457 spi驱动
- [ ] https://club.rt-thread.org/ask/article/54.html F030系列芯片UART3~UART6使用问题
- [ ] https://club.rt-thread.org/ask/question/435184.html H7 UART5的问题以及修改方法

mysterywolf, 2022-07-25:
Scons/ENV
https://github.com/RT-Thread/rt-thread/pull/6176#discussion_r927347593
https://club.rt-thread.org/ask/article/11d28badc0ddb35b.html

mysterywolf, 2022-08-02:
https://github.com/RT-Thread/packages/issues/1287 sal impl 文件夹分散出去