RT-Thread/rt-thread #5811: have wrong with rt_kprintf ("%lx") and rt_kprintf ("%llx")

GuEe-GUI, 2022-04-14:
```c
#include <stdint.h>

void _test()
{
    unsigned long ubase = 0xabcdef1234567890UL;
    uint64_t u64 = 0xabcdef1234567890UL;

    rt_kprintf("0xabcdef1234567890 %%lx = 0x%lx\n", 0xabcdef1234567890UL);
    rt_kprintf("0xabcdef1234567890 %%llx = 0x%llx\n", 0xabcdef1234567890UL);
    rt_kprintf("0xabcdef1234567890 %%lx%%lx = 0x%lx%x\n\n", 0xabcdef1234567890UL >> 32, 0xabcdef1234567890UL);

    rt_kprintf("ubase %%lx = 0x%lx\n", ubase);
    rt_kprintf("ubase %%llx = 0x%llx\n", ubase);
    rt_kprintf("ubase %%lx%%lx = 0x%lx%x\n\n", ubase >> 32, ubase);

    rt_kprintf("u64 %%lx = 0x%lx\n", u64);
    rt_kprintf("u64 %%llx = 0x%llx\n", u64);
    rt_kprintf("u64 %%lx%%lx = 0x%lx%x\n\n", u64 >> 32, u64);

    rt_kprintf("0xabcdef1234567890 %%p = 0x%p\n", u64);
}
MSH_CMD_EXPORT(_test, test)
```

RT-Thread aarch64 rt_kprintf:
```shell
msh />_test
0xabcdef1234567890 %lx = 0x34567890
0xabcdef1234567890 %llx = 0x%lx
0xabcdef1234567890 %lx%lx = 0xabcdef1234567890

ubase %lx = 0x34567890
ubase %llx = 0x%lx
ubase %lx%lx = 0xabcdef1234567890

u64 %lx = 0x34567890
u64 %llx = 0x%lx
u64 %lx%lx = 0xabcdef1234567890

0xabcdef1234567890 %p = 0xabcdef1234567890
msh />
```

Linux x64 printf:
```shell
gui@DESKTOP-J4PRI2C:~$ ./main
0xabcdef1234567890 %lx = 0xabcdef1234567890
0xabcdef1234567890 %llx = 0xabcdef1234567890
0xabcdef1234567890 %lx%lx = 0xabcdef12abcdef1234567890

ubase %lx = 0xabcdef1234567890
ubase %llx = 0xabcdef1234567890
ubase %lx%lx = 0xabcdef12abcdef1234567890

u64 %lx = 0xabcdef1234567890
u64 %llx = 0xabcdef1234567890
u64 %lx%lx = 0xabcdef12abcdef1234567890

0xabcdef1234567890 %p = 0x0xabcdef1234567890
```

%lx不能正确打印出64位的hex，需要将变量分高32位和低32位；%llx不能使用。

GuEe-GUI, 2022-04-14:
@mysterywolf

mysterywolf, 2022-04-14:
嗯 看来这个得修一下，你可以先用这个软件包打个补丁，安装上就能用：https://github.com/mysterywolf/rt_vsnprintf_full

GuEe-GUI, 2025-01-09:
@mysterywolf 如果已经修复就关闭吧

closed 2025-01-10 by mysterywolf