RT-Thread/rt-thread #5819: AT device driver layer not compatible with some LTE module with link_id zero offset is 1

esonhon, 2022-04-15:
In RTT release 4.1.0,  the file component/net/at/at_socket/at_socket.c,  function alloc_socket_by_device, the net interface id is start from zero, which can find in code line 311 "int idx = 0;", this index will transfered to at socket implementation layer, and will be passed to the AT MODULE by AT command, such as NEOWAY N58 module
"AT+TCPSETUP=<n>,<ip>,<port><CR>  <n>  链路编号，只能为 0~5。",
 but for some modules like GOSUN GM331, which says 
"AT+ZIPOPEN=<link_id>,<protocol_type>,<peer_addr>,<peer_port>[,<local_port>]  link_id>：连接标识，取值范围为[1,5] " , that means link id 0 is illegal, if we transfer 0 to module, it fails.
Do we have any straightforward and configurable way in this suitation?

xiangxistu, 2022-06-15:
Yes, we have a solution for this suitation.

https://github.com/RT-Thread/rt-thread/blob/9d0f088e200470682ee284ea3beda78bbd7a2cf0/components/net/at/at_socket/at_socket.c#L334

You can see, when this moudle have registered "socket_ops->at_socket", the "idx" will be offered by hardware driver by "AT DEVICE".

----

https://github.com/RT-Thread/rt-thread/blob/9d0f088e200470682ee284ea3beda78bbd7a2cf0/components/net/at/at_socket/at_socket.h#L74

xiangxistu, 2022-08-02:
This problem should be handled by "at_device".

closed 2022-08-02 by xiangxistu