RT-Thread/rt-thread #5841: nuclei/gd32vf103_rvstar BSP编译不通过

blta, 2022-04-19:
1. 在最新的RT-Thread master上测试
2. 使用NUclei 官方工具链riscv-nuclei-elf-gcc-10.2.0
```
> scons
scons: Reading SConscript files ...
Supported downloaded modes for board gd32vf103v_rvstar are flashxip, chosen downloaded mode is flashxip
Newlib version:4.1.0
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build\applications\main.o
riscv-nuclei-elf-gcc: error:  -march=rv32imac -mabi=ilp32 -mcmodel=medany : No such file or directory
scons: *** [build\applications\main.o] Error 1
scons: building terminated because of errors.
```
感觉像是rtconfig.py配置问题

另外使用rtthread-stdudio生成的rvstart工程，同样的编译器是可以编译通过的，
```
CC build\packages\nuclei_sdk-latest\SoC\gd32vf103\Common\Source\system_gd32vf103.o
CC libraries\gd32vf103\HAL_Drivers\drv_gpio.o
CC libraries\gd32vf103\HAL_Drivers\drv_usart.o
LINK rtthread.elf
d:/softwares/nuclei/gcc/bin/../lib/gcc/riscv-nuclei-elf/10.2.0/../../../../riscv-nuclei-elf/bin/ld.exe:E:\WorkSpaces\RT_Thread_Studio\RVSTART\packages\nuclei_sdk-latest\SoC\gd32vf103\Board\gd32vf103v_rvstar\Source\GCC\gcc_gd32vf103_flashxip.ld:69: warning: redeclaration of memory region `flash'
d:/softwares/nuclei/gcc/bin/../lib/gcc/riscv-nuclei-elf/10.2.0/../../../../riscv-nuclei-elf/bin/ld.exe:E:\WorkSpaces\RT_Thread_Studio\RVSTART\packages\nuclei_sdk-latest\SoC\gd32vf103\Board\gd32vf103v_rvstar\Source\GCC\gcc_gd32vf103_flashxip.ld:70: warning: redeclaration of memory region `ram'
riscv-nuclei-elf-objcopy -O binary rtthread.elf rtthread.bin
riscv-nuclei-elf-size rtthread.elf
   text    data     bss     dec     hex filename
 125788     796    7444  134028   20b8c rtthread.elf
scons: done building targets.
```
复制该rtconfig.py修改部分错误后编译还是同样的错误！

BernardXiong, 2022-04-19:
`riscv-nuclei-elf-gcc: error:  -march=rv32imac -mabi=ilp32 -mcmodel=medany : No such file or directory`
应该是没找到工具链，rtconfig.py中的工具链路径及配置是否正确？

blta, 2022-04-19:
> `riscv-nuclei-elf-gcc: error: -march=rv32imac -mabi=ilp32 -mcmodel=medany : No such file or directory` 应该是没找到工具链，rtconfig.py中的工具链路径及配置是否正确？

工具链配置正确的， 编译配置应该有问题，但是和rtthread-stdudio生成rvstart工程的rtconfing.py 差别不大，很奇怪！

blta, 2022-04-19:
最新进展：
```
git diff
diff --git a/bsp/nuclei/gd32vf103_rvstar/SConstruct b/bsp/nuclei/gd32vf103_rvstar/SConstruct
index 56c576acc..11e4a3ee3 100644
--- a/bsp/nuclei/gd32vf103_rvstar/SConstruct
+++ b/bsp/nuclei/gd32vf103_rvstar/SConstruct
@@ -23,7 +23,7 @@ AddOption('--run',
 DefaultEnvironment(tools=[])
 env = Environment(tools = ['mingw'],
     AS = rtconfig.AS, ASFLAGS = rtconfig.AFLAGS,
-    CC = rtconfig.CC, CFLAGS = rtconfig.CFLAGS,
+    CC = rtconfig.CC, CCFLAGS = rtconfig.CFLAGS,
     CXX = rtconfig.CXX, CXXFLAGS = rtconfig.CXXFLAGS,
     AR = rtconfig.AR, ARFLAGS = '-rc', LIBS = rtconfig.LIBS,
     LINK = rtconfig.LINK, LINKFLAGS = rtconfig.LFLAGS)

```
把Sconstruct里的CFLAG改成CCFLAG就可以了(和rtthread-stdudio生成rvstart工程保持一致)
我有空找个rvstart板子测试一下运行情况，提个PR

closed 2022-04-21 by mysterywolf