RT-Thread/rt-thread #5854: audio框架缺陷及优化方案

xhwang-China, 2022-04-21:
Hi 
在移植audio驱动时候发现，当前框架下音频record驱动只能一个，在注册第二个record时会出现assert。
调试发现是由于audio.c如下代码导致
```
--- a/components/drivers/audio/audio.c
+++ b/components/drivers/audio/audio.c
@@ -255,6 +255,7 @@ static rt_err_t _audio_dev_init(struct rt_device *dev)
     {
         struct rt_audio_record *record = (struct rt_audio_record *) rt_malloc(sizeof(struct rt_audio_record));
         rt_uint8_t *buffer;
+        char name[RT_NAME_MAX] = "";
 
         struct rt_audio_record *record = (struct rt_audio_record *) rt_malloc(sizeof(struct rt_audio_record));
         rt_uint8_t *buffer;
+        char name[RT_NAME_MAX] = "";
 
         if (record == RT_NULL)
             return -RT_ENOMEM;
@@ -268,7 +269,8 @@ static rt_err_t _audio_dev_init(struct rt_device *dev)
             LOG_E("malloc memory for for record pipe failed");
             return -RT_ENOMEM;
         }
-        rt_audio_pipe_init(&record->pipe, "record",
+        rt_snprintf(name, RT_NAME_MAX, "%s_record", audio->parent.parent.name);
+        rt_audio_pipe_init(&record->pipe, name,
                            (rt_int32_t)(RT_PIPE_FLAG_FORCE_WR | RT_PIPE_FLAG_BLOCK_RD),
                            buffer,
                            RT_AUDIO_RECORD_PIPE_SIZE);
```

这会导致多个record声卡初始化是pipe 由于名字冲突而引起的初始化失败。
个人建议系统不应该限制record声卡个数。

mysterywolf, 2022-04-21:
Hi 可以提交一个PR嘛 谢谢

xhwang-China, 2022-04-22:
Hi，抱歉，我这边公司红线不允许往github上提交任何代码。
改动记录就是上面的diff，需要麻烦你自行合入。

再次抱歉！

xhwang-China, 2022-04-22:
> Hi 可以提交一个PR嘛 谢谢

Hi，抱歉，我这边公司红线不允许往github上提交任何代码。
改动记录就是上面的diff，需要麻烦你自行合入。

再次抱歉！

Guozhanxin, 2022-04-22:
> > Hi 可以提交一个PR嘛 谢谢
> 
> Hi，抱歉，我这边公司红线不允许往github上提交任何代码。 改动记录就是上面的diff，需要麻烦你自行合入。
> 
> 再次抱歉！

如果gitee可以的话，提交到gitee上也行😜

xhwang-China, 2022-04-22:
> > > Hi 可以提交一个PR嘛 谢谢
> > 
> > 
> > Hi，抱歉，我这边公司红线不允许往github上提交任何代码。 改动记录就是上面的diff，需要麻烦你自行合入。
> > 再次抱歉！
> 
> 如果gitee可以的话，提交到gitee上也行😜

抱歉，也不行的。

mysterywolf, 2022-04-22:
好嘞 感谢 我们自己合并一下

closed 2022-06-17 by mysterywolf