RT-Thread/rt-thread #5865: STM32U5使用AT、netutils软件包时遇到硬中断错误

CN-Antonio, 2022-04-23:
开发板为NUCLEO-U575ZI-Q，MCU型号STM32U575ZITBQU。
开发环境为RT-Thread Studio 2.2.1与2.2.3，导入BSP为项目，BSP使用[stm32u575-st-nucleo](https://github.com/RT-Thread/rt-thread/tree/master/bsp/stm32/stm32u575-st-nucleo)，pull#5802，rtt版本尝试过latest, 4.1.0, 4.0.3, 4.0.2
工具链版本需选择GNU6.3.1或10.2.1

遇到的问题为：
1. 软件包中心添加AT-latest软件包，启用ESP8266，参数默认，编译通过，下载至开发板。运行后由uart1连接的msh反馈cpuport.c中rt_hw_hard_fault_exception函数，pc地址为0x08003268，查询为指向rt_strcmp函数。目前定位故障只能至此。
```
 \ | /
- RT -     Thread Operating System
 / | \     4.1.0 build Apr 23 2022 23:02:14
 2006 - 2022 Copyright by RT-Thread team
[2] I/sal.skt: Socket Abstraction Layer initialize success.
psr: 0x21000000
r00: 0x00000000
r01: 0x08025724
r02: 0x20001204
r03: 0x00000000
r04: 0x200000c8
r05: 0xdeadbeef
r06: 0xdeadbeef
r07: 0x200020b8
r08: 0xdeadbeef
r09: 0xdeadbeef
r10: 0xdeadbeef
r11: 0xdeadbeef
r12: 0xe0000000
 lr: 0x080097ff
 pc: 0x08003268
hard fault on thread: main

thread   pri  status      sp     stack size max used left tick  error
-------- ---  ------- ---------- ----------  ------  ---------- ---
tshell    20  ready   0x00000050 0x00001000    01%   0x0000000a 000
sys work  23  ready   0x00000050 0x00000800    03%   0x0000000a 000
tidle0    31  ready   0x00000054 0x00000100    32%   0x00000020 000
main      10  running 0x00000054 0x00000800    20%   0x0000000c 000
bus fault:
SCB_CFSR_BFSR:0x82 PRECISERR SCB->BFAR:00000000
```

2. 软件包添加netutils，配置使能NTP客户端，其余设置默认。编译下载后，msh输入命令ntp_sync后出现类似上述硬件中断报错。定位pc为 0x0800b416，函数ntp_get_time+298
```
 \ | /
- RT -     Thread Operating System
 / | \     4.1.0 build Apr 23 2022 23:26:00
 2006 - 2022 Copyright by RT-Thread team
[2] I/sal.skt: Socket Abstraction Layer initialize success.
msh >ntp_sync
psr: 0x41000000
r00: 0x00000000
r01: 0x20003824
r02: 0x00000072
r03: 0x00000000
r04: 0x20003838
r05: 0x08014ea8
r06: 0xdeadbeef
r07: 0x20003810
r08: 0xdeadbeef
r09: 0xdeadbeef
r10: 0xdeadbeef
r11: 0xdeadbeef
r12: 0x08015a40
 lr: 0x0800b3a9
 pc: 0x0800b416
hard fault on thread: tshell

thread   pri  status      sp     stack size max used left tick  error
-------- ---  ------- ---------- ----------  ------  ---------- ---
tshell    20  running 0x000000d4 0x00001000    12%   0x00000009 000
sys work  23  suspend 0x00000094 0x00000800    07%   0x0000000a 000
tidle0    31  ready   0x00000094 0x00000100    57%   0x0000000c 000
main      10  suspend 0x000000c8 0x00000800    20%   0x0000000c 000
bus fault:
SCB_CFSR_BFSR:0x82 PRECISERR SCB->BFAR:00000020
```

liukangcc, 2022-04-24:
单步调试一下呢

CN-Antonio, 2022-04-24:
AT-ESP8266的问题是，将断点打在main函数任何一行，开始调试，都会直接中断错误，判断最后的错误发生在rt_strcmp函数

netutils的问题是，可以进行单步调试，问题如上述所说

liukangcc, 2022-04-24:
可惜，隔离在家，没有 ESP8266，没办法帮你了

mysterywolf, 2022-04-24:
[image]
我用l475潘多拉帮你测了一下  和这两个软件包没啥关系。
应该是这个bsp本身的问题

CN-Antonio, 2022-04-24:
不需要连接ESP8266，RESET后直接中断错误，启用ESP32也会报错，只需要NUCLEO的板子就可以

liukangcc, 2022-04-24:
不添加软件包，bsp 运行有问题吗

CN-Antonio, 2022-04-24:
直接导入BSP，编译下载没有问题，红色LED闪烁，msh不报错

CN-Antonio, 2022-04-28:
这次尝试了添加AT软件包的广和通L610模组支持，同样以上问题1报错，大概是BSP存在问题。