RT-Thread/rt-thread #5884: rt-thread studio 新手上使用問題

rxin-1219, 2022-04-27:
大家好~

我剛接觸 rt-thread studio  實現第一個Hello RT-Thread! 
照著影片創一個project + 設定
第一個問題是
想連接rt-thread shell 會卡住
[image]
第二個問題是
裡用GDB 追蹤 發現卡在死迴圈

rt_thread_delay -> rt_thread_sleep -> RT_ASSERT(thread != RT_NULL) 
錯誤顯示
assertion failed at function assertion failed at function : rt_thread_sleep , line number: 538

我使用的板子是f746zg nucleo 使用TTL-uart 連接
[image]

謝謝大家

mysterywolf, 2022-04-27:
你好，请下载RT-Thread github最新源码，使用RT-Studio import（导入）的方式将https://github.com/RT-Thread/rt-thread/tree/master/bsp/stm32/stm32f746-st-nucleo 这个bsp导入到你的studio工程中

可以参见这个readme: https://github.com/RTduino/RTduino#21-%E5%B7%A5%E7%A8%8B%E7%9A%84%E5%88%9B%E5%BB%BA%E5%92%8C%E5%AF%BC%E5%85%A5

rxin-1219, 2022-04-28:
> 你好，请下载RT-Thread github最新源码，使用RT-Studio import（导入）的方式将https://github.com/RT-Thread/rt-thread/tree/master/bsp/stm32/stm32f746-st-nucleo 这个bsp导入到你的studio工程中
> 
> 可以参见这个readme: https://github.com/RTduino/RTduino#21-%E5%B7%A5%E7%A8%8B%E7%9A%84%E5%88%9B%E5%BB%BA%E5%92%8C%E5%AF%BC%E5%85%A5

嗨 Jianting

非常感謝您提供的方法
不過我在過程中還是遇到一些問題
在請教您
1.小锤子按钮編譯成功 但是沒辦法使用 Flash download 
[image]
上網找沒有相關資訊

2.rt-thread shell 依然無法使用 
我使用 ttl usb 連接至usart3(PD8 PD9)

感謝您

supperthomas, 2022-04-28:
建议发论坛看看https://club.rt-thread.org/index.html。 可以顺便把一些工程或者步骤发出来，

mysterywolf, 2022-04-28:
这是因为 你选错了下载方式 你导入工程之后默认是用的j-link, nucleo板载用的是stlink。可以在此处选择：
[image]

有问题可以在此处继续问，这个bsp用的是nucleo板载usb的虚拟串口，你直接用usb插上就可以用，不需要用ttl usb引出来一路UART，除非你有特殊用处。但是shell的输出串口默认是这个板载的usb串口奥

rxin-1219, 2022-04-29:
> 这是因为 你选错了下载方式 你导入工程之后默认是用的j-link, nucleo板载用的是stlink。可以在此处选择： [image]
> 
> 有问题可以在此处继续问，这个bsp用的是nucleo板载usb的虚拟串口，你直接用usb插上就可以用，不需要用ttl usb引出来一路UART，除非你有特殊用处。但是shell的输出串口默认是这个板载的usb串口奥

謝謝您熱心地回覆 我犯了很粗心的錯誤

想問更進階的 我想開啟timer 嘗試寫一些 動作
但是我開啟timer時會編譯會出現很多錯誤我ㄧ我
[image]
[image]
[image]

我也有使用 rt-thread evn 開啟 但也是出現同樣的錯誤
麻煩您指點

非常感謝

mysterywolf, 2022-04-29:
你好 我看了一下 这两个功能在此BSP中并未实质性配置起来，所以会出现错误。
[image]

rxin-1219, 2022-04-29:
> 你好 我看了一下 这两个功能在此BSP中并未实质性配置起来，所以会出现错误。 [image]

所以可能無法使用嗎?

mysterywolf, 2022-04-29:
对的 你需要自己通过Cubemx来进行自己适配一下。后面我会出一个视频来讲一下这块。

rxin-1219, 2022-04-29:
> 对的 你需要自己通过Cubemx来进行自己适配一下。后面我会出一个视频来讲一下这块。其实配置很简单，只需要几步就可以。

在麻煩您了 請問您的教學視頻哪裡可以觀看呢?
我想這會幫助我很多

最近因為[Neural Network on Microcontroller](https://github.com/majianjia/nnom)(NNOM) 對rt-thread 有非常大的支持
所以我想研究看看

目標是能夠跑裡面的example 
他有使用到y modem BSP中也是不支持的? (我需要傳資料給MCU)

[image]

或者可以使用另一個方法 device virtual file system 去讀取我的資料
我想要一些最基本的學習資源 不知道您能提供嗎?

supperthomas, 2022-04-29:
https://github.com/RT-Thread/rt-thread/tree/master/bsp/stm32/stm32h743-openmv-h7plus
可以用这个
https://github.com/RT-Thread/rt-thread/tree/master/bsp/stm32/stm32f746-st-nucleo
使用方法见readme.md

supperthomas, 2022-04-29:
你可以尝试把片内FLASH做成文件系统

rxin-1219, 2022-05-04:
> 有openmv的话很好 这个板子内存大 也有sd卡 https://github.com/RT-Thread/rt-thread/tree/master/bsp/stm32/stm32h743-openmv-h7plus 可以用openmv 这块板子支持的很多
嗨 我發現我的板子是 OpenMV Cam H7 R2的版本
我在使用BSP時發現
￼
不知道是不是此BSP 無法與R2版的支援
我在OPENMV IDE 使用正常
謝謝

mysterywolf, 2022-05-04:
这个BSP 是OPENMV H7 PLUS  不是 OPENMV H7

rxin-1219, 2022-05-04:
好的 謝謝 抱歉 我知道了 我的不是plus 我的是R2版本 它們長的很像 我以為只有攝像頭不一樣可以共用 另外尋文有支援h7 r2的BSP嗎 謝謝您

mysterywolf, 2022-05-04:
你要是搞神经网络什么的 其实 H7 Plus蛮适合的，32MB的内存爽到爆

rxin-1219, 2022-05-04:
主要都是CNN + RNN  想trace 微控制器運行神經網路的流程 順便學習微控制器和RTOS 的相關知識
我手上沒有H7 Plus (還是學生資金有限無法買太多設備)
只剩最後一塊 Arduino Nano 33 BLE (但我想他也是光秃秃的)

rxin-1219, 2022-05-04:
@mysterywolf 
https://www.sparkfun.com/products/18982

真的不能使用嗎
我查到他們幾乎相同
只有相機不同(我可以先不使用相機)
感謝您的回答
不好意思耽誤到妳很多時間

mysterywolf, 2022-05-04:
不可以的 这两个芯片不一样的

supperthomas, 2022-05-05:
https://github.com/RT-Thread/rt-thread/tree/master/bsp/nrf5x/nrf52840     Arduino Nano 33 BLE 可以试试这个

rxin-1219, 2022-05-05:
> https://github.com/RT-Thread/rt-thread/tree/master/bsp/nrf5x/nrf52840 Arduino Nano 33 BLE 可以试试这个

1.感謝你剛有稍微看一下readme 我身上沒有目前沒有j-link 調適器
不知道他是否可以支援USB to micro ?
2.我有另一片板子 Sipeed Maixduino  (https://reurl.cc/q552dn)
不知道有沒有支援的BSP? 

感謝

supperthomas, 2022-05-05:
> https://github.com/RT-Thread/rt-thread/tree/master/bsp/stm32/stm32f746-st-nucleo

你就用这个吧。https://github.com/RT-Thread/rt-thread/tree/master/bsp/stm32/stm32f746-st-nucleo
把flash改成文件系统就可以了。

rxin-1219, 2022-05-05:
> > https://github.com/RT-Thread/rt-thread/tree/master/bsp/stm32/stm32f746-st-nucleo
> 
> 你就用这个吧。https://github.com/RT-Thread/rt-thread/tree/master/bsp/stm32/stm32f746-st-nucleo 把flash改成文件系统就可以了。

請問如何把Flash改成文件系統
對這方面不熟悉
謝謝

rxin-1219, 2022-05-05:
我不知道打開什麼 才能找到我內部的Flash 並掛載它
[image]

supperthomas, 2022-05-05:
> 我不知道打開什麼 才能找到我內部的Flash 並掛載它 [image]

FLASH 512KB够用吗？

rxin-1219, 2022-05-05:
不確定 但想學習掛載 希望您能給我一些幫助
謝謝

supperthomas, 2022-05-05:
https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-standard/programming-manual/filesystem/filesystem?id=dfs-%e7%ae%80%e4%bb%8b

rxin-1219, 2022-05-05:
> https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-standard/programming-manual/filesystem/filesystem?id=dfs-%e7%ae%80%e4%bb%8b

需要device name 但是我不知道?
還是我可以直接call api? 但是我文件要放在哪裡?
謝謝您

rxin-1219, 2022-05-06:
@supperthomas @mysterywolf 
您好
我決定要自掏腰包買新的板子
想請問高手們有推薦我
可以學習rt-thread + NNOM 可以跑神經網路的板子嗎?
支援比較多介面
有較大的FLASH 可以掛載檔案系統

mysterywolf, 2022-05-06:
@majianjia 嘉博给个建议呗

closed 2022-05-17 by mysterywolf