RT-Thread/rt-thread #5885: at_socket问题

sidonglyl, 2022-04-27:
at_socket.c中at_recvpkt_all_delete函数感觉有问题，原代码如下
/* delete and free all receive buffer list */
static int at_recvpkt_all_delete(rt_slist_t *rlist)
{
    at_recv_pkt_t pkt = RT_NULL;
    rt_slist_t *node = RT_NULL;

    if (rt_slist_isempty(rlist))
    {
        return 0;
    }

    for(node = rt_slist_first(rlist); node; node = rt_slist_next(node))
    {
        pkt = rt_slist_entry(node, struct at_recv_pkt, list);
        if (pkt->buff)
        {
            rt_free(pkt->buff);
        }
        if (pkt)
        {
            rt_free(pkt);
            pkt = RT_NULL;
        }
    }

    return 0;
}
在第一个循环，node 指向的pkt动态空间已经被释放，然后又node = rt_slist_next(node)； 即取node->next;  next的值应该是不确定的，我觉得应该改成下面这样：
static int at_recvpkt_all_delete(slist_t *rlist)
{
    at_recv_pkt_t pkt = NULL;
    slist_t *node = NULL;

    if (rt_slist_isempty(rlist))
    {
        return 0;
    }

    for(node = rt_slist_first(rlist); node; node = rt_slist_next(rlist))
    {
        rt_slist_remove(rlist, node);
        pkt = rt_slist_entry(node, struct at_recv_pkt, list);
        if (pkt->buff)
        {
            free(pkt->buff);
        }
        if (pkt)
        {
            free(pkt);
            pkt = NULL;
        }
    }

    return 0;
}

同样的情况还出现在at_recvpkt_get函数中
/* get a block from AT socket receive buffer list */
static size_t at_recvpkt_get(rt_slist_t *rlist, char *mem, size_t len)
{
    rt_slist_t *node = RT_NULL;
    at_recv_pkt_t pkt = RT_NULL;
    size_t content_pos = 0, page_pos = 0;

    if (rt_slist_isempty(rlist))
    {
        return 0;
    }

    for (node = rt_slist_first(rlist); node; node = rt_slist_next(node))
    {
        pkt = rt_slist_entry(node, struct at_recv_pkt, list);

        page_pos = pkt->bfsz_totle - pkt->bfsz_index;

        if (page_pos >= len - content_pos)
        {
            rt_memcpy((char *) mem + content_pos, pkt->buff + pkt->bfsz_index, len - content_pos);
            pkt->bfsz_index += len - content_pos;
            if (pkt->bfsz_index == pkt->bfsz_totle)
            {
                at_recvpkt_node_delete(rlist, node);
            }
            content_pos = len;
            break;
        }
        else
        {
            rt_memcpy((char *) mem + content_pos, pkt->buff + pkt->bfsz_index, page_pos);
            content_pos += page_pos;
            pkt->bfsz_index += page_pos;
            at_recvpkt_node_delete(rlist, node);
        }
    }

    return content_pos;
}

at_recvpkt_node_delete(rlist, node);中将node释放，又进行node = rt_slist_next(node)；应当改为node = rt_slist_next(rlist);

xiangxistu, 2022-06-15:
这个问题的 PR 已经合并了：https://github.com/RT-Thread/rt-thread/pull/5971

xiangxistu, 2022-06-15:
这个 ISSUE 可以关闭了

closed 2022-06-15 by Guozhanxin