RT-Thread/rt-thread #5913: Stack Buffer Overflow Issues Report

GANGE666, 2022-05-08:
# Stack Buffer Overflow Issues Report

Commit id: `0e66f3f8337ad9790c9ac4e5c186b4da68ae92b2`

Affect version: ≤ 4.1.0

## 1. Stack Buffer Overflow in `at_gethostbyname`

In file `components/net/at/at_socket/at_socket.c` , function `at_gethostbyname`, the incoming parameter `name` should be a domain name. When a non-alphabetic string exceeding 16 bytes is encountered, a stack overflow will occur in [line 1143](https://github.com/RT-Thread/rt-thread/blob/0e66f3f8337ad9790c9ac4e5c186b4da68ae92b2/components/net/at/at_socket/at_socket.c#L1143), because the destination address `ip_str` of `strcpy` only has a capacity of 16 bytes. 

When the `name` parameter is controlled by a malicious attacker, the attacker can use this stack overflow to hijack the return address of the function, so that the attacker can execute remote code and control the entire system.

[https://github.com/RT-Thread/rt-thread/blob/0e66f3f8337ad9790c9ac4e5c186b4da68ae92b2/components/net/at/at_socket/at_socket.c#L1107-L1144](https://github.com/RT-Thread/rt-thread/blob/0e66f3f8337ad9790c9ac4e5c186b4da68ae92b2/components/net/at/at_socket/at_socket.c#L1107-L1144)

## 2. Stack Buffer Overflow in `at_getaddrinfo`

In file `components/net/at/at_socket/at_socket.c` , function `at_getaddrinfo` , the incoming parameter `nodename` should be a domain name. When a non-alphabetic string exceeding 16 bytes is encountered, a stack overflow will occur in [line 1251](https://github.com/RT-Thread/rt-thread/blob/0e66f3f8337ad9790c9ac4e5c186b4da68ae92b2/components/net/at/at_socket/at_socket.c#L1251), because the destination address `ip_str` of `strcpy` only has a capacity of 16 bytes. 

When the `nodename` parameter is controlled by a malicious attacker, the attacker can use this stack overflow to hijack the return address of the function, so that the attacker can execute remote code and control the entire system.

[https://github.com/RT-Thread/rt-thread/blob/0e66f3f8337ad9790c9ac4e5c186b4da68ae92b2/components/net/at/at_socket/at_socket.c#L1171-L1181](https://github.com/RT-Thread/rt-thread/blob/0e66f3f8337ad9790c9ac4e5c186b4da68ae92b2/components/net/at/at_socket/at_socket.c#L1171-L1181)

[https://github.com/RT-Thread/rt-thread/blob/0e66f3f8337ad9790c9ac4e5c186b4da68ae92b2/components/net/at/at_socket/at_socket.c#L1235-L1253](https://github.com/RT-Thread/rt-thread/blob/0e66f3f8337ad9790c9ac4e5c186b4da68ae92b2/components/net/at/at_socket/at_socket.c#L1235-L1253)

Please check the overflow issues above and add the corresponding constraints if necessary.

xiangxistu, 2022-08-02:
yeah, maybe we shouldn't handle the fixed length for characters for "ip_str".   the memory aboult “ip_str” is belong thread stack, it would cause overflow by unnormal parameter.
-------

When code choose branch "else", it means "nodename" and “name” both are ip address.   whatever the address is ipv4 or ipv6, the memory of “ip_str”is enough to store it.

GANGE666, 2022-08-02:
> Yeah, maybe we shouldn't handle the fixed length for characters for "ip_str". the memory aboult “ip_str” is belong thread stack, it would cause overflow by unnormal parameter.
>
> When code choose [...]

Thanks for checking.
I think you should fix case 1 "ip_str" overflow due to the users or developers of RT-Thread will not be aware that the parameter to this function `at_gethostbyname` is limited in length.

If you choose to fix this bug, you can refer to the corresponding commit in this issue, and I am happy to confirm the validity of the patch.

lazyparser, 2026-10-14:
This thread has been quiet for over two years. From the discussion, the overflow risk around ip_str handling in at_gethostbyname still looks open (or not?). Could we settle on a direction — bound the copy, or reject over-long hostnames — and have someone own a fix? If it has been addressed elsewhere, noting the commit here would let this be closed.