RT-Thread/rt-thread #5939: 定时器组件的问题(timer.c)

slyant, 2022-05-13:
相关代码：
`void rt_timer_init(rt_timer_t  timer,
                   const char *name,
                   void (*timeout)(void *parameter),
                   void       *parameter,
                   rt_tick_t   time,
                   rt_uint8_t  flag)
{
    /* parameter check */
    RT_ASSERT(timer != RT_NULL);
    RT_ASSERT(timeout != RT_NULL);
    RT_ASSERT(time < RT_TICK_MAX / 2);

    /* timer object initialization */
    rt_object_init(&(timer->parent), RT_Object_Class_Timer, name);

    _timer_init(timer, timeout, parameter, time, flag);
}`
和
`rt_timer_t rt_timer_create(const char *name,
                           void (*timeout)(void *parameter),
                           void       *parameter,
                           rt_tick_t   time,
                           rt_uint8_t  flag)
{
    struct rt_timer *timer;

    /* parameter check */
    RT_ASSERT(timeout != RT_NULL);
    RT_ASSERT(time < RT_TICK_MAX / 2);

    /* allocate a object */
    timer = (struct rt_timer *)rt_object_allocate(RT_Object_Class_Timer, name);
    if (timer == RT_NULL)
    {
        return RT_NULL;
    }

    _timer_init(timer, timeout, parameter, time, flag);

    return timer;
}`

RT_ASSERT(time < RT_TICK_MAX / 2);这个参数的处理建议不要这么“狠”，断言无异于“杀头”。很多时候物联设备需要远程配置定时参数，并保存到设备，如果不小心保存了错误的值（time > RT_TICK_MAX / 2），造成设备一启动就断言失败死机，连远程升级固件的机会都没有了，这是灾难！！！建议只做警告，然后修剪值为<RT_TICK_MAX / 2，或者返回创建失败。

Guozhanxin, 2022-05-13:
感谢反馈，我们看看这个问题

wineee, 2022-05-20:
同意这个观点，应该返回创建失败，生产环境尽量避免断言

mysterywolf, 2022-05-20:
这块建议应该重新处理一下assert的使用时机，
如果开启rt_assert，说明目前是处于调试模式，而不是大规模产品部署，可以直接卡死来提醒工程师这块在编程上是有问题的。
如果关闭rt_assert，应该在检查值出现问题之后直接范围RT_ERROR或者其他错误码，让程序动态的处理。