RT-Thread/rt-thread #5941: [BSP][ESP32-C3] task issue

supperthomas, 2022-05-13:
task issue as follow

- [ ] combine IDF and scons --targe=cmake build system
- [ ] porting drv_gpio.c with cmake system
- [ ] porting drv_uart.c with cmake system
- [ ] porting drv_iic.c with cmake system
- [ ] porting drv_spi.c with cmake system
- [ ] porting drv_pwm.c with cmake system

supperthomas, 2022-07-09:
fix IDF_PATH to local path  目前可以考虑将IDF_PATH路径修改成pkgs里面的软件包作为编译的时候的IDF_PATH. 不过尝试多次之后，效果不是很理想，也比较麻烦。 暂时不是很方便，后面再考虑，先把软件包提交上去。如果大伙有什么更好的办法，欢迎提供。目前需求是： 如何编译的时候将IDF_PATH 指定到本地package/rtt_esp_idf-latest 文件夹下面，而且可以顺利编译通过。

supperthomas, 2022-09-14:
#6247

supperthomas, 2022-09-16:
新版ESP32_C3 BSP发布 
https://club.rt-thread.org/ask/article/46de5889adf71acc.html

supperthomas, 2022-09-16:
【ESP32C3】基于ESP32C3运行RT-THREAD  

https://club.rt-thread.org/ask/article/6749217c538b42cd.html

supperthomas, 2022-12-06:
【请社区公测】新版ESP32_C3 BSP发布
https://club.rt-thread.org/ask/article/46de5889adf71acc.html

supperthomas, 2023-10-31:
@WCX1024979076
单独打开BLE 这个app_main没有定义。
https://github.com/RT-Thread/rt-thread/actions/runs/6709048939/job/18231225581

```
LINK rtthread.elf
/opt/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/11.2.0/../../../../riscv32-esp-elf/bin/ld: build/main/main.o: in function `main':
/home/runner/work/rt-thread/rt-thread/bsp/ESP32_C3/main/main.c:24: undefined reference to `app_main'
```

WCX1024979076, 2023-10-31:
这个需要参考readme里面将esp-idf的nimble样例程序加入到编译中

 

 https://github.com/RT-Thread/rt-thread/actions/runs/6709048939/job/18231225581

supperthomas, 2023-10-31:
> 这个需要参考readme里面将esp-idf的nimble样例程序加入到编译中
> […](#)

[image]

这里用户自己操作起来问题会比较多，是不是可以提供一个step by step的步骤？

supperthomas, 2023-11-01:
@WCX1024979076 
https://club.rt-thread.org/ask/article/419a30e57384a239.html
可否看下能不能默认状态下打开BLE宏，编译能通过？

WCX1024979076, 2023-11-01:
好的，我看一下

 

 https://club.rt-thread.org/ask/article/419a30e57384a239.html

1078249029, 2024-08-29:
请问现在的工具链可以确保编译成功么？

我在使用esp官方的gcc编译链产生编译错误：
```
scons: Reading SConscript files ...
CROSS_TOOL= gcc
EXEC_PATH= /home/lzx/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build/drivers/board.o
riscv32-esp-elf-gcc: fatal error: cannot read spec file 'nosys.specs': No such file or directory
compilation terminated.
scons: *** [build/drivers/board.o] Error 1
scons: building terminated because of errors.
```
然后我在rtconfig.py文件中指定nosys.specs位置
`DEVICE  = ' -nostartfiles -march=rv32imc --specs=/usr/lib/riscv32-esp-elf/riscv32-esp-elf/nosys.specs '`
之后又缺失cc1工具，当时使用脚本下载乐鑫的sdk被默认下载到/usr/lib里了，然后我又将cc1复制到了乐鑫默认sdk文件夹里` cp /usr/lib/riscv32-esp-elf/libexec/gcc/riscv32-esp-elf/11.2.0/cc1 /home/lzx/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin`
然后又爆出标准库文件丢失的错误
```
scons: Reading SConscript files ...
CROSS_TOOL= gcc
EXEC_PATH= /home/lzx/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build/drivers/board.o
In file included from drivers/board.c:12:
packages/ESP-IDF-latest/components/hal/include/hal/systimer_hal.h:9:10: fatal error: stdint.h: No such file or directory
    9 | #include <stdint.h>
      |          ^~~~~~~~~~
compilation terminated.
scons: *** [build/drivers/board.o] Error 1
scons: building terminated because of errors.
```
根据网上的信息**很可能是工具链设置出错**，但是我是根据目前最新的教程移植的，请问楼主您最近移植rtt到esp32c3了么？可以编译通过么？

下面是我的**环境配置**

gcc：
```
lzx@DESKTOP-EPJK785 ESP32_C3 (master) $ /home/lzx/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc -v
Using built-in specs.
COLLECT_GCC=/home/lzx/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
Target: riscv32-esp-elf
Configured with: /builds/idf/crosstool-NG/.build/riscv32-esp-elf/src/gcc/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=riscv32-esp-elf --prefix=/builds/idf/crosstool-NG/builds/riscv32-esp-elf --exec_prefix=/builds/idf/crosstool-NG/builds/riscv32-esp-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/riscv32-esp-elf/riscv32-esp-elf --with-headers=/builds/idf/crosstool-NG/builds/riscv32-esp-elf/riscv32-esp-elf/include --with-newlib --enable-threads=no --disable-shared --with-arch=rv32gc --with-abi=ilp32 --with-pkgversion='crosstool-NG esp-13.2.0_20230928' --disable-__cxa_atexit --enable-cxx-flags=-ffunction-sections --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --disable-libstdcxx-verbose --with-gmp=/builds/idf/crosstool-NG/.build/riscv32-esp-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/riscv32-esp-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/riscv32-esp-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/riscv32-esp-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-nls --enable-multiarch --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio --enable-libstdcxx-time=yes
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 13.2.0 (crosstool-NG esp-13.2.0_20230928)
```

rtconfig.py
```
import os

# toolchains options
ARCH        ='risc-v'
CPU         =''
CROSS_TOOL  ='gcc'

if os.getenv('RTT_CC'):
    CROSS_TOOL = os.getenv('RTT_CC')
print("CROSS_TOOL=",CROSS_TOOL)

if CROSS_TOOL == 'gcc':
    PLATFORM    = 'gcc'
    EXEC_PATH   = r'~/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin'
#    EXEC_PATH   = r'~/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/'  # 三套编译链都试过了，后两套找不到ll库
#    EXEC_PATH   = r'~/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin'
[... 7 lines cut ...]
print("EXEC_PATH=",EXEC_PATH)
BUILD = 'debug'

if PLATFORM == 'gcc':
    # toolchains
    PREFIX = 'riscv32-esp-elf-'
    CC = PREFIX + 'gcc'
    CXX = PREFIX + 'g++'
    AS = PREFIX + 'gcc'
    AR = PREFIX + 'ar'
    LINK = PREFIX + 'g++'
    TARGET_EXT = 'elf'
    SIZE = PREFIX + 'size'
    OBJDUMP = PREFIX + 'objdump'
    OBJCPY = PREFIX + 'objcopy'
    STRIP = PREFIX + 'strip'

    DEVICE  = ' -nostartfiles -march=rv32imc --specs=/usr/lib/riscv32-esp-elf/riscv32-esp-elf/nosys.specs '
    CFLAGS = DEVICE + '-gdwarf-4 -ggdb -Og'
    AFLAGS =  ' -c' + DEVICE + ' -x assembler-with-cpp'
    LFLAGS = DEVICE + ' -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32C3=0 -Wl,--gc-sections,-Map=rtthread.map,-cref,-u,_start -T idf_port/ld/memory.ld -T idf_port/ld/sections.ld -T packages/ESP-IDF-latest/components/esp_rom/esp32c3/ld/esp32c3.rom.ld -T packages/ESP-IDF-latest/components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld -T packages/ESP-IDF-latest/components/esp_rom/esp32c3/ld/esp32c3.rom.libgcc.ld -T packages/ESP-IDF-latest/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib.ld -T packages/ESP-IDF-latest/components/esp_rom/esp32c3/ld/esp32c3.rom.version.ld -T packages/ESP-IDF-latest/components/esp_rom/esp32c3/ld/esp32c3.rom.eco3.ld -T packages/ESP-IDF-latest/components/soc/esp32c3/ld/esp32c3.peripherals.ld  -Wl,--wrap=_Unwind_SetEnableExceptionFdeSorting -Wl,--wrap=__register_frame_info_bases -Wl,--wrap=__register_frame_info -Wl,--wrap=__register_frame -Wl,--wrap=__register_frame_info_table_bases -Wl,--wrap=__re [... cut ...] d_GetLanguageSpecificData -Wl,--wrap=_Unwind_FindEnclosingFunction -Wl,--wrap=_Unwind_Resume -Wl,--wrap=_Unwind_RaiseException -Wl,--wrap=_Unwind_DeleteException -Wl,--wrap=_Unwind_ForcedUnwind -Wl,--wrap=_Unwind_Resume_or_Rethrow -Wl,--wrap=_Unwind_Backtrace -Wl,--wrap=__cxa_call_unexpected -Wl,--eh-frame-hdr -Wl,--wrap=__gxx_personality_v0  -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32C3=0'
    CXXFLAGS = CFLAGS

    POST_ACTION = OBJCPY + ' -Oihex $TARGET rtthread.hex\n' + SIZE + ' $TARGET \n'
```

supperthomas, 2024-08-31:
> 请问现在的工具链可以确保编译成功么？
>
> 我在使用esp官方的gcc编译链产生编译错误：
>
> ```
> scons: Reading SConscript files ...
> CROSS_TOOL= gcc
> EXEC_PATH= /home/lzx/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin [...]

https://github.com/RT-Thread/rt-thread/actions/runs/10645398186/job/29511156571
请参考这个链接

1078249029, 2024-09-03:
> https://github.com/RT-Thread/rt-thread/actions/runs/10645398186/job/29511156571 请参考这个链接

谢谢，根据action已经配好了。应该是我之前下载esp-if工具链冲突了。后期我看看能不能把spi和pwm的驱动给补上

supperthomas, 2024-09-03:
ok, 感谢反馈

1078249029, 2024-10-08:
这是我写的spi驱动，大佬看看还有什么问题？[PR](https://github.com/RT-Thread/rt-thread/pull/9513)。 @supperthomas

supperthomas, 2025-02-26:
感谢大家，esp32c3支持已经完成