RT-Thread/rt-thread #5946: RT-Thread操作系统贡献指南

Guozhanxin, 2022-05-15:
# RT-Thread操作系统贡献指南

欢迎来到开源摘星计划-RT-Thread操作系统贡献专栏 。RT-Thread操作系统已加入开源摘星计划，该计划提供奖励以鼓励你加入我们的社区：所有参与本项目贡献的开发者都可获得相应贡献积分、奖励以及被评选为年度摘星100 人的机会，活动及奖励详情请见：[开源摘星计划](https://github.com/weopenprojects/WeOpen-Star)。

📅活动时间：2022年5月-12月

## 🌟关于RT-Thread操作系统

RT-Thread诞生于2006年，是一款以开源、中立、社区化发展起来的物联网操作系统。 RT-Thread主要采用 C 语言编写，浅显易懂，且具有方便移植的特性（可快速移植到多种主流 MCU 及模组芯片上）。RT-Thread把面向对象的设计方法应用到实时系统设计中，使得代码风格优雅、架构清晰、系统模块化并且可裁剪性非常好。

RT-Thread有完整版和Nano版，对于资源受限的微控制器（MCU）系统，可通过简单易用的工具，裁剪出仅需要 3KB Flash、1.2KB RAM 内存资源的 NANO 内核版本；而相对资源丰富的物联网设备，可使用RT-Thread完整版，通过在线的软件包管理工具，配合系统配置工具实现直观快速的模块化裁剪，并且可以无缝地导入丰富的软件功能包，实现类似 Android 的图形界面及触摸滑动效果、智能语音交互效果等复杂功能。

RT-Thread是一个集实时操作系统（RTOS）内核、中间件组件的物联网操作系统，架构如下：

[image]

### **在开始贡献之前：**

- 请加入【开源摘星计划】社群

  请添加官方小助手微信（微信号：TCOSOO1）

### **以下是项目可参与的方向：**

- 测试

测试是最简单的贡献途径，可以以[提交issue](https://github.com/RT-Thread/rt-thread/issues)的形式进行反馈，在任何一个新版本、新软件或者新功能上线都需要进行多种测试保证功能能稳定运行。如果您刚开始进行贡献，不妨从测试入手。

- 提交Issue/解决已有Issue

可以以[提交issue](https://github.com/RT-Thread/rt-thread/issues)的形式进行反馈使用过程中的问题，同时可以在针对不同的issue进行修复，每个issue下面都有参与者的讨论，欢迎您发表您的看法

- 软件拓展建议

如果您在使用RT-Thread中途发现有软件功能的缺失，可以以[提交issue](https://github.com/RT-Thread/rt-thread/issues)的形式进行反馈。

- 贡献代码/工具

如果您想为RT-Thread开发中间件或者其他工具，可以以[提交issue](https://github.com/RT-Thread/rt-thread/issues)或加入底部“开源PR交流群”的形式进行反馈，开发完成后可以参考[向 RT-Thread 贡献代码](https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-standard/development-guide/github/github)这篇文档提交代码到仓库中。或参考[软件包开发指南](https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-standard/development-guide/package/package?id=软件包开发指南)以软件包的形式提交。

### 🌟我们对贡献任务进行了分类，以帮助你找到你感兴趣的任务

### **社区日常任务**

社区日常任务
积分奖励：1分/个

- 提交 Issue/bug

文档类
积分奖励：10分/个

- 完善RT-Thread社区文档里面的已有文档[英文文档](https://github.com/RT-Thread/rt-thread/tree/master/documentation)、[中文文档](https://gitee.com/rtthread/docs-online)
- 提交与RT-Thread社区相关的新的文档

编程类
积分奖励：20分~50分/个

- 修复已有issue
- 开发适配RT-Thread操作系统中间件、工具等

其他类

- 开源知识分享官：提交对本项目的学习笔记或开源相关笔记（视频/文章皆可）并发布在任意社区/博客/视频平台即可申报积分。
  积分奖励：10分
- 开源部署测试：安装部署RT-Thread操作系统，截图使用界面即可申报积分。
  积分奖励：10分
- 额外奖励：每月将为10名优秀笔记作者奖励QQ周边公仔、QQ音乐/腾讯视频年卡及腾源会T恤等好礼。

## **贡献帮助**

官方微信公众号：RTThread物联网操作系统

贡献交流群：

[image]

对于成功完成任意任务的贡献者，为记录你的积分以及发放奖励，请自行前往你的积分主页记录。积分记录规则详见：[关于积分](https://github.com/weopenprojects/WeOpen-Star/tree/main/earn-points.md)。

blta, 2022-05-15:
这个有意思！

ZIL520, 2022-06-27:
nice

closed 2023-03-06 by mysterywolf