RT-Thread/rt-thread #5953: CROSS_TOOL 选择交叉编译工具的问题

thewon86, 2022-05-16:
```
# toolchains options
CROSS_TOOL 	= 'gcc'

if os.getenv('RTT_CC'):
	CROSS_TOOL = os.getenv('RTT_CC')
```
在所有的 bsp 里面 rtconfig.py 几乎都有这么两行代码，前面告诉使用者通过修改 CROSS_TOOL 来选择 gcc keil iar 等等编译工具。
后面两句的 if 语句是什么意思？

如下修改后，发现 print 打印出来的不是 keil ，是 gcc ！
```
CROSS_TOOL='keil'

if os.getenv('RTT_CC'):
    CROSS_TOOL = os.getenv('RTT_CC')
print(CROSS_TOOL)
```

前边需要用户定制编译工具的，后面那个处理是什么操作，为啥有 RTT_CC 这个环境变量就用环境变量的值？这个环境变量定义的值是多少？默认环境变量定义的是gcc？

Guozhanxin, 2022-05-16:
是的，我也感觉这段代码有问题，没啥好方法。win下的env工具里因为有内置的工具链，所以会设定RTT_CC和RTT_EXEC_PATH这两个环境变量。默认使用gcc编译。

BernardXiong, 2022-05-16:
```python
# toolchains options
CROSS_TOOL 	= 'gcc'

if os.getenv('RTT_CC'):
	CROSS_TOOL = os.getenv('RTT_CC')
```

CROSS_TOOL默认使用gcc；当环境变量中设置了RTT_CC时，则使用环境变量中的设置。

场景，例如在Windows下可能存在gcc，keil mdk，iar时，可以通过环境变量来选择不同的工具链，在不同的工具链间进行切换。

thewon86, 2022-05-17:
> ```python
> # toolchains options
> CROSS_TOOL 	= 'gcc'
> 
> if os.getenv('RTT_CC'):
> 	CROSS_TOOL = os.getenv('RTT_CC')
> ```
> 
> CROSS_TOOL默认使用gcc；当环境变量中设置了RTT_CC时，则使用环境变量中的设置。
> 
> 场景，例如在Windows下可能存在gcc，keil mdk，iar时，可以通过环境变量来选择不同的工具链，在不同的工具链间进行切换。

问题就在环境变量中的 RTT_CC 不是通过系统属性-环境变量添加进去的。而是 env 环境代码里添加的。好像是在 cmdinit.cmd 中用 set 语句设置的。

wugensheng-code, 2022-05-24:
应该是可以通过设置环境变量来设置的，可以在CMD或者powershell下试一下

closed 2022-06-12 by mysterywolf