RT-Thread/rt-thread #5978: at socket在远端断开连接时的问题

NU-LL, 2022-05-21:
在at_closed_notice_cb中，当断开时会修改当前状态：

https://github.com/RT-Thread/rt-thread/blob/2fc7fab80a9b3d67632e1a94598519a788716aac/components/net/at/at_socket/at_socket.c#L676-L691

所以，当远端断开链接后，at_recvfrom中，下面这里之前是不是应该判断下当前socket的状态？

https://github.com/RT-Thread/rt-thread/blob/2fc7fab80a9b3d67632e1a94598519a788716aac/components/net/at/at_socket/at_socket.c#L832-L840

即加入：

```c
if(sock->state == AT_SOCKET_CLOSED)
{
    result = -1;
    goto __exit;
}
```

不太确定是不是应该这样修改，如果是的话我这边可以提交下PR

目前这样修改后，在应用程序中通过recv是能够正常获取到远端断开链接的消息

xiangxistu, 2022-06-15:
recv 这里应该有 CLOSED 状态的参与，但是如果这样加的话，可能就没有办法接收对端已经发来的数据了。

----

这种方式在前些版本时是有这样的逻辑，不过因为上面的原因删掉了。

-----

在使用中出现 close_notice_cb 调整了这个 at socket 的连接状态；并且会唤醒处于等待状态的 recv ，如果有数据就接收数据，一直接收到没有数据；如果没有数据，就基本可以判定是 close_notice_cb 来唤醒的，返回错误。
可以看看这个逻辑：https://github.com/RT-Thread/rt-thread/pull/6024/files#diff-b9c9ab9940e9ce9b2fc5abfc838ea7e41d52c227b4cdfb888db3d688470548ed

NU-LL, 2022-06-15:
是的 这样修改的话应该也可以解决这个close状态的问题

xiangxistu, 2022-06-16:
https://github.com/RT-Thread/rt-thread/pull/6024

closed 2022-06-16 by xiangxistu