RT-Thread/rt-thread #5979: qemu-riscv-virt64 smode 下无法启动

wineee, 2022-05-21:
使用工具：
QEMU emulator version 7.0.0
opensbi v1.0

目前 m态 运行正常，但是 S态 无法启动：

./qemu-nographic-smode.sh 执行结果：
Exception:
Load access fault
--------------Dump Registers-----------------
Function Registers:
xcause = 00000005,xtval = 80208386,xepc = 80224d00

Exception:
Load access fault
--------------Dump Registers-----------------
Function Registers:
xcause = 00000005,xtval = 80208386,xepc = 80224d00

Exception:
Load access fault
--------------Dump Registers-----------------
Function Registers:
xcause = 00000005,xtval = 80208386,xepc = 80224d00

Exception:
Load access fault
--------------Dump Registers-----------------
Function Registers:
xcause = 00000005,xtval = 80208386,xepc = 80224d00
无限循环了

我是用 scons --pyconfig 配置的：
[image]

大概测试了一下历史版本 https://github.com/R T-Thread/rt-thread/commit/cda8caeda0cf105dd660192e367017eea0d407ec 是可以用的

wineee, 2022-05-21:
(gdb) s
sbi_set_timer (val=6897667) at driver/sbi.c:125
125         if (has_time_extension)
=> 0x0000000080201dc4 <sbi_set_timer+12>:       97 77 02 00     auipc   a5,0x27
   0x0000000080201dc8 <sbi_set_timer+16>:       93 87 47 ec     addi    a5,a5,-316 # 0x80228c88 <has_time_extension>
   0x0000000080201dcc <sbi_set_timer+20>:       83 c7 07 00     lbu     a5,0(a5)
(gdb) s
132             (void)SBI_CALL1(SBI_SET_TIMER, 0, val);
=> 0x0000000080201e18 <sbi_set_timer+96>:       01 48   li      a6,0
   0x0000000080201e1a <sbi_set_timer+98>:       81 47   li      a5,0
   0x0000000080201e1c <sbi_set_timer+100>:      01 47   li      a4,0
   0x0000000080201e1e <sbi_set_timer+102>:      81 46   li      a3,0
   0x0000000080201e20 <sbi_set_timer+104>:      03 36 84 fd     ld      a2,-40(s0)
   0x0000000080201e24 <sbi_set_timer+108>:      81 45   li      a1,0
   0x0000000080201e26 <sbi_set_timer+110>:      01 45   li      a0,0
   0x0000000080201e28 <sbi_set_timer+112>:      ef f0 9f f1     jal     ra,0x80201d40 <sbi_call>
(gdb) s
sbi_call (arg7=arg7@entry=0, arg6=arg6@entry=0, arg0=arg0@entry=6897667, 
    arg1=arg1@entry=0, arg2=arg2@entry=0, arg3=arg3@entry=0, arg4=arg4@entry=0)
    at driver/sbi.h:150
150         register uintptr_t a0 __asm("a0") = (uintptr_t)(arg0);
=> 0x0000000080201d62 <sbi_call+34>:    03 35 84 fb     ld      a0,-72(s0)
(gdb) s
151         register uintptr_t a1 __asm("a1") = (uintptr_t)(arg1);
=> 0x0000000080201d66 <sbi_call+38>:    83 35 04 fb     ld      a1,-80(s0)
(gdb) s
152         register uintptr_t a2 __asm("a2") = (uintptr_t)(arg2);
=> 0x0000000080201d6a <sbi_call+42>:    03 36 84 fa     ld      a2,-88(s0)
(gdb) s
153         register uintptr_t a3 __asm("a3") = (uintptr_t)(arg3);
=> 0x0000000080201d6e <sbi_call+46>:    83 36 04 fa     ld      a3,-96(s0)
(gdb) s
154         register uintptr_t a4 __asm("a4") = (uintptr_t)(arg4);
=> 0x0000000080201d72 <sbi_call+50>:    03 37 84 f9     ld      a4,-104(s0)
(gdb) s
155         register uintptr_t a6 __asm("a6") = (uintptr_t)(arg6);
=> 0x0000000080201d76 <sbi_call+54>:    03 38 04 fc     ld      a6,-64(s0)
(gdb) s
156         register uintptr_t a7 __asm("a7") = (uintptr_t)(arg7);
=> 0x0000000080201d7a <sbi_call+58>:    83 38 84 fc     ld      a7,-56(s0)
(gdb) s
158         __asm __volatile(\
=> 0x0000000080201d7e <sbi_call+62>:    73 00 00 00     ecall
(gdb) s
0x0000000080000408 in ?? ()
=> 0x0000000080000408:  73 12 02 34     csrrw   tp,mscratch,tp
(gdb) s
Cannot find bounds of current function
(gdb) s
Cannot find bounds of current function

wineee, 2022-05-21:
目前debug到这了，明天我再看看

supperthomas, 2022-05-22:
欢迎尝试修正，PR。

FatPigeorz, 2022-11-10:
在https://github.com/RT-Thread/rt-thread/commit/557821fd3dfb6e13fc26b65e199508ba09f49872 中和的最近的commit https://github.com/RT-Thread/rt-thread/commit/f57eb4feb5ee4856299058058eff15593ac0570d 下都没发现这种情况
[image]
[image]

mysterywolf, 2023-03-11:
solve in here: https://github.com/RT-Thread/rt-thread/pull/7040

closed 2023-03-11 by mysterywolf