RT-Thread/rt-thread #5980: Posix信号量destrory机制问题

CornerOfSkyline, 2022-05-21:
在Posix IPC中的semaphore的API `sem_destroy()`在实现过程中会在销毁信号量变量前会尝试获取该信号量，以避免删除正在使用的信号量，但当信号量初始值为0，存在以下内存泄漏风险：

[image]

在上图所示场景中，`TaskA`创建用于同步的信号量`sync_sem`，初值为0，随后`TaskA`使`TaskB`就绪（`TaskB`优先级低于`TaskA`）,`TaskA`通过等待`sync_sem`挂起自己，`TaskB`运行并完成任务后会释放`sync_sem`，`TaskA`在获得`sync_sem`后被激活并删除`sync_sem`。

**查看信号量相关步骤实现细节：**

**第1步：`sem_init()`**

> https://github.com/RT-Thread/rt-thread/blob/2fc7fab80a9b3d67632e1a94598519a788716aac/components/libc/posix/ipc/semaphore.c#L180-L211
> 其中会调用`rt_sem_create()`来创建rtt的信号量，并将信号量的**value**设置为**0**
> https://github.com/RT-Thread/rt-thread/blob/2fc7fab80a9b3d67632e1a94598519a788716aac/src/ipc.c#L408

**第3步：`sem_wait()`**

> https://github.com/RT-Thread/rt-thread/blob/2fc7fab80a9b3d67632e1a94598519a788716aac/components/libc/posix/ipc/semaphore.c#L372-L390
> 其中会调用`rt_sem_take()`来获取信号量或者挂起等待，因`sync_sem`的**value**值为**0**，因此会将`TaskA`添加进`sync_sem`的**挂起队列**，`sync_sem`的**value**仍保持为**0**
> https://github.com/RT-Thread/rt-thread/blob/2fc7fab80a9b3d67632e1a94598519a788716aac/src/ipc.c#L503-L539

**第5步：`sem_post()`**

> https://github.com/RT-Thread/rt-thread/blob/2fc7fab80a9b3d67632e1a94598519a788716aac/components/libc/posix/ipc/semaphore.c#L297-L315
> 其中会调用`rt_sem_release()`来获取释放信号量，其中会先判断`sync_sem`的**挂起队列**是否为空，如果不为空则恢复挂起队列里的第一个进程，并启动调度器，因`sync_sem`的**挂起队列**中存在`TaskA`，所以会恢复`TaskA`，保持信号量的**value**值**0**
> https://github.com/RT-Thread/rt-thread/blob/2fc7fab80a9b3d67632e1a94598519a788716aac/src/ipc.c#L627-L645

**第6步：`sem_destory()`**

> https://github.com/RT-Thread/rt-thread/blob/2fc7fab80a9b3d67632e1a94598519a788716aac/components/libc/posix/ipc/semaphore.c#L108-L135
> 其中首先会调用`rt_sem_trytake()`去尝试获取`sync_sem`信号量，`rt_sem_trytake()`通过调用无等待的`rt_sem_take()`来实现
> https://github.com/RT-Thread/rt-thread/blob/2fc7fab80a9b3d67632e1a94598519a788716aac/src/ipc.c#L587-L590
> 在`rt_sem_take()`中，由于`sync_sem`的**value**值为**0**，会**返回超时错误**
> https://github.com/RT-Thread/rt-thread/blob/2fc7fab80a9b3d67632e1a94598519a788716aac/src/ipc.c#L503-L539
> https://github.com/RT-Thread/rt-thread/blob/2fc7fab80a9b3d67632e1a94598519a788716aac/src/ipc.c#L514-L519
> `rt_sem_trytake()`**返回超时错误后**，`sem_destroy()`会设置错误码为`EBUSY`并返回，导致`sync_sem`在内存中的实体未能删除
> https://github.com/RT-Thread/rt-thread/blob/2fc7fab80a9b3d67632e1a94598519a788716aac/components/libc/posix/ipc/semaphore.c#L121-L128

CornerOfSkyline, 2022-05-21:
该使用场景是存在合理需求的，使用信号量来进行线程间同步，发现此问题后尝试通过在`sem_destroy`前调用一次`sem_post`使信号量的**value值大于0**以使`rt_sem_trytake`可以返回成功，但会导致执行至下列代码时由于**地址为0**引发Hardfault或者提示该区域正在使用断言错误
https://github.com/RT-Thread/rt-thread/blob/2fc7fab80a9b3d67632e1a94598519a788716aac/src/mem.c#L147

thewon86, 2022-05-23:
既然是 destroy ，为啥要去 take 呢？

CornerOfSkyline, 2022-05-27:
> 既然是 destroy ，为啥要去 take 呢？

应该是为了避免删掉正在使用的信号量 @thewon86

BernardXiong, 2022-06-05:
> 在Posix IPC中的semaphore的API `sem_destroy()`在实现过程中会在销毁信号量变量前会尝试获取该信号量，以避免删除正在使用的信号量，但当信号量初始值为0，存在以下内存泄漏风险：
>
> [image] [...]

赞一个，非常详尽。请问下，这个问题在RT-Thread semaphore中是否存在？后续关于POSIX IPC应该需要统一修一波，同时追加更多的测试用例。

xiangxistu, 2022-06-15:
```sem_destory``` 的语义就是删除信号量：
删除该信号量应该有以下几种情况：

1. 有线程在使用该 sem ：
   1. 允许等待就等待，使用 rt_sem_take() 以有等待时间的方式等待 sem 释放
   2. 不允许等待但允许强制删除就直接删除，删除前设置等待该 sem 的线程为就绪态。
   3. 不允许等待但也不允许强制删除就返回错误。
2. 没有线程在使用该 sem:
   1. 直接删除

----

无论是 trytake 还是 take 均不能实现发现是否有其他线程在使用该 sem 。

----

在 POSIX 中也没有明说应该具体怎么做，只有这样的表述：
```
It is safe to destroy an initialized semaphore upon which no threads are currently blocked. The effect of destroying a semaphore upon which other threads are currently blocked is undefined.
```
在 value 这里的 trytake 确实没有看到有什么实际的意义。因为 posix 中也没有定义初始化时候的 value必须要大于 0。
```
The value of the initialized semaphore shall be value.
```

closed 2022-08-01 by mysterywolf