RT-Thread/rt-thread #5991: rt_device_create()

gw495817787, 2022-05-24:
创建设备附加user_data时，rt_device_create()只为user_data分配了空间，但是并没有对user_data指针进行初始化，导致使用user_data数据时读写异常，需要额外再设备驱动初始化时对其进行赋值，造成一定的困扰。

mysterywolf, 2022-05-24:
谢谢反馈

mysterywolf, 2022-05-24:
顺便请问一下，你是在使用了rt-thread设备框架的情况下，在设备框架和驱动之外自己在应用层赋值了user_data嘛？

BernardXiong, 2022-05-26:
这个指的是什么？代码中，device_create时会分配结构体内存，并会进行清零。

gw495817787, 2022-05-27:
> 顺便请问一下，你是在使用了rt-thread设备框架的情况下，在设备框架和驱动之外自己在应用层赋值了user_data嘛？

是的，按字面理解，这个user_data是用户数据。在应用中赋值时失败。查看内核源代码，发现rt_device_create设备时，虽然分配和清零相关内存块，但是device->user_data这个指针并没有指向attach_size的这块内存。 

rt_device_t rt_device_create(int type, int attach_size)
{
    int size;
    rt_device_t device;

    size = RT_ALIGN(sizeof(struct rt_device), RT_ALIGN_SIZE);
    attach_size = RT_ALIGN(attach_size, RT_ALIGN_SIZE);
    /* use the total size */
    size += attach_size;

    device = (rt_device_t)rt_malloc(size);
    if (device)
    {
        rt_memset(device, 0x0, sizeof(struct rt_device));
        device->type = (enum rt_device_class_type)type;
    }

    return device;
}
RTM_EXPORT(rt_device_create);

ufbycd, 2024-08-11:
本人初次用rt-thread，看到`rt_device_create`的内部实现，也是很奇怪。
这个接口似乎都没人用，是这个issue导致的吗？话说这么久了，这个issue还没修正呀？！

Rbb666, 2024-12-15:
[image]
rt_device_create用法可以参照上面所示使用

closed 2024-12-15 by Rbb666