RT-Thread/rt-thread #5999: [bsp][qemu-vexpress-a9] 文件系统有问题

supperthomas, 2022-05-26:
试了下qemu，文件系统有问题
[image]

BernardXiong, 2022-05-28:
啊，其他现象描述呢？fatfs？是否mount 上了？

supperthomas, 2022-05-29:
> 

没有，sd0 没有mount上。

supperthomas, 2022-05-29:
> 啊，其他现象描述呢？fatfs？是否mount 上了？

这个好像是时序的问题，等有空来解一下，或者其他小伙伴有兴趣解一下。应该很简单。

supperthomas, 2022-05-29:
root cause: rt_hw_us_delay 这个函数没有实现，需要实现一下：
```
rt_hw_us_delay() doesn't support for this board.Please consider implementing rt_hw_us_delay() in another file.rt_hw_us_delay() doesn't support for this board.Please consider implementing rt_hw_us_delay() in another file.rt_hw_us_delay() doesn't support for this board.Please consider implementing rt_hw_us_delay() in another file.Hello RT-Thread!

```

chenyingchun0312, 2022-06-01:
尝试再次mount，同样失败

[image]

supperthomas, 2022-06-01:
第一次使用，如果list_device中sd0没有mount上，就是文件系统不能用，进入rtthread执行如下命令

mkfs -t elm sd0

closed 2022-06-01 by mysterywolf

mysterywolf, 2022-06-08:
如果文档中心有问题的话，可以直接改。