RT-Thread/rt-thread #6007: 关于rtt驱动文档的建议

ccx-lan, 2022-05-28:
目前rtt似乎没有底层驱动对接的相关文档，遇到一些没有驱动的bsp，想要对接也只能参考一些已有的芯片bsp，但对于一些比较复杂的驱动比如sdio，usb等对于小白来说还是比较麻烦的，希望大佬们能写一些文档，比如需要基本实现哪些接口等等

mysterywolf, 2022-05-28:
谢谢建议 这确实是痛点 还有就是设备框架原理对于初学者也是门槛

key-star, 2022-05-31:
我觉得设备框架应该为上层应用提供统一的api，但是也有看到应用层通过调用`rt_pin_mode`，`rt_spi_transfer`等接口直接驱动设备，在使用时候就会有疑惑，使用`rt_device_write`和`rt_pin_write`有什么区别，所以我感觉在芯片资源足够的情况下应该使用统一的api，为资源受限设备提供另外api，并通过文档进行说明

mysterywolf, 2022-05-31:
https://club.rt-thread.org/ask/question/231413344ffb4032.html

mysterywolf, 2022-05-31:
> 我觉得设备框架应该为上层应用提供统一的api，但是也有看到应用层通过调用`rt_pin_mode`，`rt_spi_transfer`等接口直接驱动设备，在使用时候就会有疑惑，使用`rt_device_write`和`rt_pin_write`有什么区别，所以我感觉在芯片资源足够的情况下应该使用统一的api，为资源受限设备提供另外api，并通过文档进行说明

感谢反馈 之前还真没想过有人会在这个地方纠结，后续会在文档和视频中做出说明，我先简单和你解释一下，rt_device_write这种是所有设备的统一接口，rt_pin_write是针对gpio的操作接口，你可以理解为他比rt_device_write的目标更明确也更易用。rt_spi_transfer也是一样的。换句话说，rt_device_write/read由于需要统一所有设备，所以他的接口对于某些设备来说比较诡异，比如write函数中的一些参数根本就没有用到，因此rt_pin_write相当于一个gpio专用版的device_write，用着更舒服也更易用。你直接用device_write去操作gpio也是没有问题的。

BernardXiong, 2022-07-27:
这部分也包括，可能rt_device_read/write等API对于一些外设、硬件是不容易覆盖的，这个时候用另外形式的API会更合适，使用上更容易

closed 2022-08-02 by mysterywolf