RT-Thread/rt-thread #6021: 关于drv_eth.c编译报错

key-star, 2022-06-02:
基于stm32f429-atk-apollo，RT-Thread版本为v4.1.0，在移植lwip时，遇到如下几个问题：
1.drv_eth.c和phy.h中的两个enum中的元素PHY_FULL_DUPLEX重复定义，论坛上有人提问，但是没有很好的回答；
2.drv_eth.c无法驱动cubeMX生成的最新的库文件，最新的库文件与bsp当前在用的库文件对比出来有较大差异，drv_eth.c是否有必要适配最新的库文件。

xiangxistu, 2022-08-03:
[image]
---
f429-atk-apollo 在使用 cubemx(version 6.6.1) + STM32F4_Hal(version 1.27.1) 使用时，按照本身的 sconscirpt 管理逻辑，不会出现 drv_eth.c, phy.h 的冲突问题。不会发生的原因是，没有使用到 CubeMX 提供的新的 library 而是继续沿用 /stm32/libraries/STM32F4xxx_HAL 。也就是说，CubeMX生成的新版本的 Hal 库本身是可以不参与编译的。
----

这个问题，可以与 /stm32/libraries/STM32Fxxxx 的库更新绑定在一起，看看是否有必要随着 CubeMX一直更新。

xiangxistu, 2022-08-03:
至于 phy.h 中的定义，我感觉可以删掉了；phy 种类比较多，这个类型确实没必要

closed 2022-08-04 by mysterywolf