RT-Thread/rt-thread #6025: [Linux & English] Can IDE run natively on Linux?

rtek1000, 2022-06-02:
Hello,

Can IDE run natively on Linux?

Because I tried some compilers for ARM and all of them compiled much faster when using Linux instead of Windows.

---

It would be nice for more people to read if people write the issues in English, translation to other languages ​​is better if it's from English.

---

I came here because I researched about RTOS compiler for Arm920T (S3C2440).

Thanks for sharing!

https://community.infineon.com/t5/Legacy-microcontrollers/What-RTOS-support-is-there-for-S6E2CC-Series/td-p/45504?start=0&tstart=0

supperthomas, 2022-06-02:
Hi @rtek1000 :
you can follow this document to build the environment for linux. 

https://www.rt-thread.io/document/site/documentation/quick_start_qemu/quick_start_qemu_linux/

BYW, there's other document for you to know the RTOS for this link
https://www.rt-thread.io/document/site/introduction/introduction/

mysterywolf, 2022-06-02:
Hi, Here is the latest version: https://github.com/RT-Thread/rt-thread/blob/master/documentation/quick-start/quick_start_qemu/quick_start_qemu_linux.md
If you would like to use MAC OS: here is the tutorial: https://github.com/RT-Thread/rt-thread/blob/master/documentation/quick-start/quick_start_qemu/quick_start_qemu_macos.md

Thank you for you suggestion. But we cannot force every community number to use English, which is a non-native language, to chat and discuss with others. Google translate may overcome this problem.

BernardXiong, 2022-06-05:
Maybe we can open Pandora's box and use vscode + smart extension to develop RT-Thread under Linux.

rtek1000, 2022-06-05:
ST and NPX manufacturers make use of Eclipse:

[image]

[image]

rtek1000, 2022-06-05:
There is also another IDE called ChibiStudio, which also has an Eclipse distribution:

> The best way to get started is to download ChibiStudio, an Eclipse-based environment with everything required in order to evaluate ChibiOS. 

https://osdn.net/projects/chibios/

[image]

closed 2022-06-21 by mysterywolf