RT-Thread/rt-thread #6106: drv_usart.c的dma_isr中的recv_len处理失当。

SnowOfSpring, 2022-06-23:
static void dma_isr(struct rt_serial_device *serial)
{
    struct stm32_uart *uart;
    rt_size_t recv_total_index, recv_len;
    rt_base_t level;

    RT_ASSERT(serial != RT_NULL);
    uart = rt_container_of(serial, struct stm32_uart, serial);

    if ((__HAL_DMA_GET_IT_SOURCE(&(uart->dma_rx.handle), DMA_IT_TC) != RESET) ||
            (__HAL_DMA_GET_IT_SOURCE(&(uart->dma_rx.handle), DMA_IT_HT) != RESET))
    {
        level = rt_hw_interrupt_disable();
        recv_total_index = serial->config.bufsz - __HAL_DMA_GET_COUNTER(&(uart->dma_rx.handle));
        if (recv_total_index == 0)
        {
            recv_len = serial->config.bufsz - uart->dma_rx.last_index;
        }
        else
        {
            // recv_len = recv_total_index - uart->dma_rx.last_index;  
            // 触发中断时recv_total_index 肯定为零，但当代码执行到此时，可能 recv_total_index > 0
            // 而如果此时 recv_total_index < uart->dma_rx.last_index时，recv_len将为负

           // 如果不考虑 recv_total_index >= uart->dma_rx.last_index 数据覆盖的情况，此处应更正为：
            recv_len = serial->config.bufsz - uart->dma_rx.last_index +  recv_total_index;
        }
        uart->dma_rx.last_index = recv_total_index;
        rt_hw_interrupt_enable(level);

        if (recv_len)
        {
            rt_hw_serial_isr(serial, RT_SERIAL_EVENT_RX_DMADONE | (recv_len << 8));
        }
    }
}

thewon86, 2022-06-24:
https://gitee.com/thewon/serialX
这种逻辑鬼才问题，在 serialX 里不存在

SnowOfSpring, 2022-06-27:
> https://gitee.com/thewon/serialX 这种逻辑鬼才问题，在 serialX 里不存在

if ((__HAL_DMA_GET_IT_SOURCE(&(uart->dma_rx.handle), DMA_IT_TC) != RESET) ||
        (__HAL_DMA_GET_IT_SOURCE(&(uart->dma_rx.handle), DMA_IT_HT) != RESET))
    {
        **dma_cnt = RT_SERIAL_DMA_BUFSZ - __HAL_DMA_GET_COUNTER(&(uart->dma_rx.handle));**

        rt_hw_serial_isr(serial, RT_SERIAL_EVENT_RX_DMADONE | (dma_cnt << 8));
    }

确实如你所写，完全没必要进行if (recv_total_index == 0)判断，可以在Idle中断中后续处理增长的字节。

nongxiaoming, 2022-07-26:
DMA_IT_TC的时候会是0，但是如果是DMA_IT_HT的时候不是0吧

mysterywolf, 2022-08-04:
https://github.com/RT-Thread/rt-thread/issues/4467#issuecomment-850814143

mysterywolf, 2022-08-04:
感谢反馈 已在此处修复 https://github.com/RT-Thread/rt-thread/pull/6226

closed 2022-08-05 by mysterywolf