RT-Thread/rt-thread #6107: [bsp问题]stm32f469-discovery 开发板触摸屏驱动报错

piaoxuebingfeng, 2022-06-24:
在使用 stm32f469-discovery 开发板运行bsp包中的 lvgl 示例的时候，发现触摸屏驱动不支持，串口报错日志如下：

```
[I/I2C] I2C bus [i2c1] registered
[D/drv.i2c] software simulation i2c1 init done, pin scl: 24, pin sda 25

 \ | /
- RT -     Thread Operating System
 / | \     4.1.0 build Jun 22 2022 21:45:11
 2006 - 2022 Copyright by RT-Thread team
[I/SDIO] SD card capacity 61069312 KB.
found part[0], begin: 1048576, size: 58.245GB
[I/sal.skt] Socket Abstraction Layer initialize success.
[I/at.clnt] AT client(V1.3.1) on device uart6 initialize success.
msh />[E/TOUCH.ft] ft_read i2c read error: -8
[E/TOUCH.ft] ft_probe failed: -1
[E/TOUCH] no touch screen or do not have driver
```

list_device 发现i2c1 设备已经初始化，但是没有使用的设备，触摸屏驱动报错。

查看bsp的说明，已经适配了对应的触摸屏驱动了。

请问有什么解决方案？

snoooopy, 2022-08-10:
我现在遇到了和你一样的问题，触摸芯片是FT6206，你解决这个问题了没？

snoooopy, 2022-08-11:
我用了最新的rt-thread源码，配置时候选择了跟我板子上一样的6206，但是还是会报跟你上面一样的错，不知道为什么，用其他的程序驱动触摸这块没有任何问题

mysterywolf, 2022-08-11:
这两种不同型号的芯片的板子我们都是实际测试下载过的了

snoooopy, 2022-08-11:
因为我昨天是用的scons导出的cmake，然后用clion编译的，我现在试一下用rt-thread studio直接创建一个469-disc的工程，先测试一下看看是否是之前menuconfig配置的不对。

snoooopy, 2022-08-11:
我心态有点崩，用rt-thread studio创建工程编译各种报错，找不到头文件，明明头文件可以定向找到，编译就是报错，放弃了studio，然后用mdk，编译后仍然是触摸iic驱动报错
 \ | /
- RT -     Thread Operating System
 / | \     4.1.1 build Aug 11 2022 10:03:03
 2006 - 2022 Copyright by RT-Thread team
msh >[E/TOUCH.ft] ft_read i2c read error: -8
[E/TOUCH.ft] ft_probe failed: -1
[E/touch] no touch screen or do not have driver
触摸芯片已经选择了FT6202还是不行，我再试试吧

snoooopy, 2022-08-12:
在ft_probe这个函数中，用ft_read去读FT6206的A8地址时候返回的是false，读不回芯片id，我直接暴力方法屏蔽这段代码，ft_probe直接返回true，触摸驱动正常了，这块代码是有些问题吧，具体我还没看到底啥问题。

mysterywolf, 2022-08-12:
你看一下读出来的芯片id是什么？ 顺便可否用放大镜看一下 这个芯片的具体型号？我怎么感觉这有三种不同的触摸芯片在f469上呢😂

mysterywolf, 2022-08-12:
@liYony 麻烦看一下这个问题

snoooopy, 2022-08-12:
芯片id读回来的是0xFF

mysterywolf, 2022-08-12:
https://github.com/RT-Thread-packages/ft6206

mysterywolf, 2022-08-12:
你看一下 把这个改回来试试 https://github.com/RT-Thread-packages/ft6206/commit/296fa257165f4e418820478bcb38da1dbcf55a38

snoooopy, 2022-08-12:
芯片型号是FT6206GMA

liYony, 2022-08-12:
你看看触摸选的是不是ft6206

snoooopy, 2022-08-12:
> 你看看触摸选的是不是ft6206
> […](#)

是的，我用menuconfig配置了FT6202，程序中的驱动部分也都是基于6202的，这个应该没错

snoooopy, 2022-08-12:
> https://github.com/RT-Thread-packages/ft6206

这个驱动单独用没有问题，我测试过，可以读到触摸的位置信息，初始化也都是正常的，但是这个驱动中没有用到读A8芯片ID的，所以可以用，然后我用st官方的程序看了一下读芯片ID，读回来的的确也是0x11，也是对的。不知为什么就在ft_probe中读不回id

liYony, 2022-08-12:
你先自己看一下，我们当时测试是好使的。

  
https://github.com/RT-Thread-packages/ft6206

mysterywolf, 2022-08-19:
请问读不回ID指的是什么，应该不会阻塞住啊，至少应该返回个0xff啥的吧

mysterywolf, 2022-08-22:
@snoooopy 请问这个问题还存在吗?

closed 2022-09-05 by mysterywolf

liYony, 2022-10-11:
请问你是哪一款触摸芯片？