RT-Thread/rt-thread #6110: EEPROM的设备框架

xfwangqiang, 2022-06-26:
从源码上看，rt-thread还没有一个EEPROM的设备框架，是否考虑新增这个。关于memory的设备目前已经有了设备类型RT_Device_Class_MTD和RT_Device_Class_Block，但从这两个设备的API看来，主要还是面向Flash型的设备。

supperthomas, 2022-06-27:
可以先尝试提交个软件包上来，我们一起看看。

xfwangqiang, 2022-06-27:
因为涉及到设备类型，需要增加RT_Device_Class_EEPROM，可能按软件包来估计不行，要不我先提PR？

supperthomas, 2022-06-27:
> 因为涉及到设备类型，需要增加RT_Device_Class_EEPROM，可能按软件包来估计不行，要不我先提PR？

EEPROM 是不是通过IIC通信的？可否用这个RT_Device_Class_I2CBUS ?  你可以尝试整理下AT24Cxx的API。

ccx-lan, 2022-06-28:
也有一些eeprom是spi接口的，不过还是i2c接口的多

supperthomas, 2022-06-28:
> 也有一些eeprom是spi接口的，不过还是i2c接口的多

可以搞个软件包，对接SPI 或者I2C，上层操作应该差不多的吧？提供接口。

xfwangqiang, 2022-06-28:
> 

如果放在软件包，从驱动往EEPROM的设备框架层注册时，就只能用现有的设备枚举类型，用RT_Device_Class_I2CBUS或是RT_Device_Class_SPI应该都不合适，RT_Device_Class_MTD和RT_Device_Class_Block也好像不合适，
好像RT_Device_Class_MTD比其它几个会合适点，那就用这个吧

supperthomas, 2022-06-30:
> > 
> 
> 如果放在软件包，从驱动往EEPROM的设备框架层注册时，就只能用现有的设备枚举类型，用RT_Device_Class_I2CBUS或是RT_Device_Class_SPI应该都不合适，RT_Device_Class_MTD和RT_Device_Class_Block也好像不合适， 好像RT_Device_Class_MTD比其它几个会合适点，那就用这个吧

可以先用这个RT_Device_Class_Miscellaneous

mysterywolf, 2022-08-11:
你好 EEPROM 我们这里有一份NASA 的EEFS文件系统，是否可以考虑作为框架？ 美国航天局用在卫星上的：我只是整理了一下，但是还没有适配到rt-thread上来 https://github.com/RT-Thread-packages/eefs 如果感兴趣 我可以给你权限。

xfwangqiang, 2022-08-12:
> 你好 EEPROM 我们这里有一份NASA 的EEFS文件系统，是否可以考虑作为框架？ 美国航天局用在卫星上的：我只是整理了一下，但是还没有适配到rt-thread上来 https://github.com/RT-Thread-packages/eefs 如果感兴趣 我可以给你权限。

我看了下这个仓库，通过readme了解来看，这个应该是个文件系统，应该跟FatFS或LitteFS一样，只要是个MTD的设备就可以挂载，是不是应该跟EEPROM解耦才好？

mysterywolf, 2022-08-13:
是的！是得先有个EEPROM设备框架才行