RT-Thread/rt-thread #6112: nxp1052 串口bsp驱动，如果应用程序对某个串口重新配置波特率等参数，该串口无法接收数据，需将中断重新使能

chunyexixiaoyu, 2022-06-27:
应用程序如果重新配置串口参数会调用
`static rt_err_t imxrt_configure(struct rt_serial_device *serial, struct serial_configure *cfg)
{
    struct imxrt_uart *uart;
    lpuart_config_t config;

    RT_ASSERT(serial != RT_NULL);
    RT_ASSERT(cfg != RT_NULL);

    uart = rt_container_of(serial, struct imxrt_uart, serial);

    LPUART_GetDefaultConfig(&config);
    config.baudRate_Bps = cfg->baud_rate;

    switch (cfg->data_bits)
    {
    case DATA_BITS_7:
        config.dataBitsCount = kLPUART_SevenDataBits;
        break;

    default:
        config.dataBitsCount = kLPUART_EightDataBits;
        break;
    }

    switch (cfg->stop_bits)
    {
    case STOP_BITS_2:
        config.stopBitCount = kLPUART_TwoStopBit;
        break;
    default:
        config.stopBitCount = kLPUART_OneStopBit;
        break;
    }

    switch (cfg->parity)
    {
    case PARITY_ODD:
        config.parityMode = kLPUART_ParityOdd;
        break;
    case PARITY_EVEN:
        config.parityMode = kLPUART_ParityEven;
        break;
    default:
        config.parityMode = kLPUART_ParityDisabled;
        break;
    }

    config.enableTx = true;
    config.enableRx = true;

    LPUART_Init(uart->uart_base, &config, GetUartSrcFreq(uart->uart_base));

    return RT_EOK;
}`
但是调用后串口无法收到数据，需再次中断使能
`        LPUART_EnableInterrupts(uart->uart_base, kLPUART_RxDataRegFullInterruptEnable);
        NVIC_SetPriority(uart->irqn, NVIC_EncodePriority(NVIC_GetPriorityGrouping(), 4, 0));
        EnableIRQ(uart->irqn);`

nongxiaoming, 2022-07-26:
一般需要在open之前进行波特率配置，配置完了再open的。串口驱动开关中断是在control里面进行的，open的时候框架会根据你的flag进行开关中断操作。如果配置就提前开启中断可能会导致串口在没有open的时候就触发中断导致断言死机。

chunyexixiaoyu, 2022-07-26:
> 一般需要在open之前进行波特率配置，配置完了再open的。串口驱动开关中断是在control里面进行的，open的时候框架会根据你的flag进行开关中断操作。如果配置就提前开启中断可能会导致串口在没有open的时候就触发中断导致断言死机。

但是如果posix接口 通过ioctl这种操作就有点问题的。

mysterywolf, 2022-08-09:
> 但是如果posix接口 通过ioctl这种操作就有点问题的。

ioctl串口的这个问题4.1.1会解决