RT-Thread/rt-thread #6132: 关于 rt_tick_increase 函数在调度器启动前被调用可能导致空指针访问的问题。

sszllzss, 2022-07-01:
rt_tick_increase 函数如下，我们一般在rt_hw_board_init 中调用SysTick_Config 来启动滴答定时器，当还未启动调度器(rt_system_scheduler_start)之前如果发送滴答定时器中断(SysTick_Handler)调用rt_tick_increase，此时rt_thread_self函数返回NULL，在操作thread->remaining_tick，将发送数据总线错误。
目前我的解决办法是，在rt_system_scheduler_start添加一个钩子函数，来启动滴答定时器，请问还有更好的解决办法吗？（个人不行因为这个问题在rt_tick_increase中添加判断，会影响系统的CPU消耗）
```
void rt_tick_increase(void)
{
    struct rt_thread *thread;
    rt_base_t level;

    level = rt_hw_interrupt_disable();

    /* increase the global tick */
#ifdef RT_USING_SMP
    rt_cpu_self()->tick++;
#else
    ++rt_tick;
#endif /* RT_USING_SMP */

    /* check time slice */
    thread = rt_thread_self();

    --thread->remaining_tick;
    if (thread->remaining_tick == 0) {
        /* change to initialized tick */
        thread->remaining_tick = thread->init_tick;
        thread->stat |= RT_THREAD_STAT_YIELD;

        rt_hw_interrupt_enable(level);
        rt_schedule();
    } else {
        rt_hw_interrupt_enable(level);
    }

    /* check timer */
    rt_timer_check();
}
```

jhbdream, 2022-07-08:
cpu 全局中断在开始调度之前是关闭的，当开始调度第一个任务时才会打开全局中断 此时才会进入 tick 中断

sszllzss, 2022-07-09:
这个后面发现了，这个问题是因为我BootLoader打开了滴答定时器，跳转到APP后汇编中开启了全局中断，可能导致rt_tick_increase被调用

closed 2022-07-09 by sszllzss