RT-Thread/rt-thread #6138: 反馈GD32 BSP编译bug

CharlesXSong, 2022-07-04:
BSP路径：[rt-thread](https://github.com/RT-Thread/rt-thread)/[bsp](https://github.com/RT-Thread/rt-thread/tree/master/bsp)/[gd32](https://github.com/RT-Thread/rt-thread/tree/master/bsp/gd32)/[arm](https://github.com/RT-Thread/rt-thread/tree/master/bsp/gd32/arm)/[gd32305r-start](https://github.com/RT-Thread/rt-thread/tree/master/bsp/gd32/arm/gd32305r-start)
错误文件：rt-thread\components\libc\compilers\armlibc\syscalls.c
错误行数：line(21)
错误内容：..\..\..\components\libc\compilers\armlibc\syscalls.c(21): error:  #5: cannot open source input file "fcntl.h": No such file or directory

暂时解决办法：将`#include <fcntl.h>`改为`#include "fcntl\octal\fcntl.h"`

CharlesXSong, 2022-07-04:
拉错源了……是`gitee_master`的代码😶

closed 2022-07-04 by CharlesXSong

CharlesXSong, 2022-07-04:
然而master里的工程还是编译不了……提示一大堆`#20: identifier "va_list" is undefined`

mysterywolf, 2022-07-04:
你现在拉下来的是开发版本，工程是需要手动刷新的，因为我们的仓库在开发中，代码随时在变化。

CharlesXSong, 2022-07-05:
> 你需要用scons --target=mdk5 重新生成一下工程

感谢，原来是`RT_USING_LIBC`没有定义，重新scons一下就在mdk工程文件里生成了；BTW，v4.1.0已发行版本这个BSP的GPIO驱动有问题，没有定义GPIOE，所以我才来拉最新版的，看到master仓库里已经优化过了，小坑已踩🤭